wjwwood/serial #88: Build fail. Deps are installed. On arch linux machine

SlightlyCyborg, 2015-01-25:
`[cyborg@evo-van serial]$ make
cd build && cmake -DCMAKE_INSTALL_PREFIX=/tmp/usr/local ..
CMake Error at CMakeLists.txt:5 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/cyborg/programs/dd/DailyData-RoomDoorSensor/dep/serial/build/CMakeFiles/CMakeOutput.log".
Makefile:35: recipe for target 'serial' failed
make: *** [serial] Error 1
`

wjwwood, 2015-01-25:
You don't have catkin installed or it isn't on your `CMAKE_PREFIX_PATH`, according to that error message.

You can install catkin or you can create a minimal CMake project to build and install serial as described in this comment:

https://github.com/wjwwood/serial/issues/52#issuecomment-28848337

wjwwood, 2015-01-25:
Also, if you resolve the issue please follow up and close this issue, thanks.

closed 2015-04-22 by wjwwood