wjwwood/serial #89: serialConfig.cmake not found

costas290, 2015-02-27:
I was able to successfully build and install `serial` but I can't include it in my CMake project.

This is the error I get:

```
CMake Warning at src/CMakeLists.txt:21 (find_package):
  By not providing "Findserial.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "serial", but
  CMake did not find one.

  Could not find a package configuration file provided by "serial" with any
  of the following names:

    serialConfig.cmake
    serial-config.cmake

  Add the installation prefix of "serial" to CMAKE_PREFIX_PATH or set
  "serial_DIR" to a directory containing one of the above files.  If "serial"
  provides a separate development package or SDK, be sure it has been
  installed.
```

I believe it's because `serial` itself doesn't provide a package configuration file. Is there any solution to this problem?

wjwwood, 2015-02-27:
Did you make sure that place you installed serial to is on your `CMAKE_PREFIX_PATH`?

If you installed it with `catkin`, there should be a `setup.bash` file you can source (`source /path/to/install/setup.bash`) which will setup your environment so that other CMake projects can find it.

wjwwood, 2015-04-22:
Closing this for now, please comment with more info.

closed 2015-04-22 by wjwwood

stephanepoirier, 2017-01-13:
I encountered a similar issue when following installation instructions from the README. As mentioned by @wjwwood , that was a `CMAKE_PREFIX_PATH` issue.

Any of these worked for me:
- set `CMAKE_PREFIX_PATH` to `/tmp/usr/local/setup.sh` (default _serial_ installation directory) in the CMake project that uses _serial_
- build _serial_ with CMake, and leave its `CMAKE_INSTALL_PREFIX` to `/usr/local` (default), which is on the `CMAKE_SYSTEM_PREFIX_PATH`. Therefore, the resulting Makefile installs _serial_ to a location visible to all CMake projects.

plguhur, 2018-03-08:
In my case, I solved this issue by sourcing `/tmp/usr/local/setup.sh` as suggested by @wjwwood and @stephanepoirier.

aarontan94, 2018-05-15:
sudo apt-get install ros-indigo-serial

fixed it for me

dltmdwls25, 2018-05-20:
ㄱㅅㄱㅅ

itzranjeet, 2020-12-11:
sudo apt-get install ros-melodic-serial
worked for me.

amjack0, 2021-12-01:
I have the same issue with noetic ?

shrutichakraborty, 2024-06-26:
hello, I have this issue on ros2 humble, what would you suggest?