wjwwood/serial #94: Jade release?

mikepurvis, 2015-04-22:
Got some drivers blocked on this.

wjwwood, 2015-04-22:
Ok, let me just audit the pull requests and issues real quick and I'll get a release out.

Also, I was considering converting this to a plain CMake package for Jade, do you foresee that being an issue? Maybe I should send a message to ros-users's about it.

mikepurvis, 2015-04-22:
It makes it a pain in the butt to hack on it in conjunction with a catkin package, though that's less of a big deal these days what with `catkin_tools` and all. It would be helpful if it was pulled in by one of the metapackages, such that it wouldn't be a non-catkin thing needing to be added on top of the base ROS install.

wjwwood, 2015-04-22:
> It makes it a pain in the butt to hack on it in conjunction with a catkin package, though that's less of a big deal these days what with catkin_tools and all.

I was afraid of that, unfortunately there's no way to get it to work with `catkin_make`, I'll keep that as a consideration.

> It would be helpful if it was pulled in by one of the metapackages, such that it wouldn't be a non-catkin thing needing to be added on top of the base ROS install.

I don't quite follow this, which metapackage should it go under? I'm not aware of any sticking points in the development workflow because this is not a part of any official ROS variants. Maybe you could explain more?

To be clear I still planned on releasing it through the ROS build farm, not rolling my own debs.

mikepurvis, 2015-04-22:
Oh yes, and it's very little change on Ubuntu— I'm thinking of platforms like Arch and OS X where the basis of a ROS install is a gigantic wstool invocation and an hour-long build of everything.

As it stands, if I want to hack on the um6 or roboteq drivers on my Mac, I end up just building serial in the same workspace. I can still do that; I'll just have to switch to catkin tools.

wjwwood, 2015-04-22:
I'm not going to change the build system right now. I might do it post Jade in a new version like 1.3.0 and look at K turtle for releasing that version into ROS. I want to try, between now and then, to make a hybrid that uses catkin if it is there, but falls back to plain CMake otherwise. We'll see if I actually take the time to do it.

I'm doing the Jade release now.

wjwwood, 2015-04-22:
https://github.com/ros/rosdistro/pull/8038

closed 2015-04-22 by wjwwood