wjwwood/serial #100: Conversion error

lordninjamelon, 2015-06-01:
I have been trying to send out encoded protocol buffer data, but I keep getting conversion errors. I am using https://github.com/nanopb/nanopb library, but I do not think that is the issue.

This is code snippet from where the compile finds errors:

```c++
OdroidToArduino messageOut;
messageOut.spd = 200;
messageOut.angle = 20;
uint8_t bufferOut[64];
pb_ostream_t streamOut = pb_ostream_from_buffer((uint8_t*)bufferOut, sizeof(bufferOut));

bool sendStatus = pb_encode(&streamOut, OdroidToArduino_fields, &messageOut);
if (sendStatus) {
   for (int i = 0; i < 10; i++) {
   my_serial.write(bufferOut[i]);
}
```

This is the error  I get
[image]

wjwwood, 2015-06-01:
Hi, you're using the wrong version of the write method, if you're passing a single character then you need to provide a size. By passing one argument the compiler is trying to use the `write(std::string &)` method.

You can probably fix this by doing something like this:

```c++
my_serial.write(&bufferOut[i]);
```

Please comment if this doesn't resolve the issue.

wjwwood, 2015-06-01:
Also, in the future, please include the error text not a screenshot.

closed 2015-06-01 by wjwwood

lordninjamelon, 2015-06-01:
Thank you for the quick reply, unfortunately I am still having problems

This is the error I get with your suggested change

```
/usr/include/c++/4.8/bits/basic_string.tcc:212:5: note:   no known conversion for argument 1 from ‘uint8_t* {aka unsigned char*}’ to ‘const char*’
/home/dundua/Desktop/2015-mini-smart-vehicles-master/project-template/sources/OpenDaVINCI-msv/apps/proxy/src/Proxy.cpp:228:40: error: invalid conversion from ‘uint8_t* {aka unsigned char*}’ to ‘const char*’ [-fpermissive]
           my_serial.write(&bufferOut[i]);
```

wjwwood, 2015-06-01:
Ah, it still can't convert the single `char` to a `std::string`, try with the explicit size:

```c++
my_serial.write(&bufferOut[i], 1);
```

lordninjamelon, 2015-06-02:
That fixed it, thank you