wjwwood/serial #101: Serial::setTimeout should accept param by const reference

theidealist, 2015-06-03:
It's misleading (and inconsistent with the constructor version of the same API) to accept this parameter by non-const reference as it indicates to me the user that it is being "captured" by reference and that I should therefore keep the timeout object I provide it around for at least as long as I have the Serial object itself, or that the object will be modified by the function itself. Neither of these do I believe to be true after closer inspection of the code.

If there is data being changed in Timeout (I followed the pimpl implementation through to unix.cc:702) it should be advertised more clearly in the documentation.

Thanks!

wjwwood, 2015-06-08:
I agree, the `setTimeout` method should take a `const Timeout &`. The constructor should probably do the same. I just need to test how the default argument works out in that case.

I'll do this when I get some time, but it would go quicker with a pull request. Also, I'll need to consider how this breaks the ABI/API and then figure out what future release it should be included into.

theidealist, 2015-06-15:
I submitted PR 102 to hopefully make this go a little faster.  I hope it is acceptable.

Thanks for your attention @wjwwood !