wjwwood/serial #108: cannot build

ElectricRCAircraftGuy, 2015-08-25:
Please forgive my ignorance. I am good on Arduino/Atmel AVR, but have never done C or C++ programming on a computer (vs microcontroller). 

I want to run your serial example to start. I am using Windows 8 with CodeBlocks, GNU GCC Compiler. I downloaded your library zip file, extracted it, navigated there using command prompt, and typed, make, and I got the following:

[image]

I can't figure out how to install and use the library to run your test file....

wjwwood, 2015-08-25:
The `Makefile` isn't designed with Windows in mind. It looks like `sh` crashed because `fork` is unavailable? Unfortunately, I have no idea how to fix that. I recommend you use the included VS file on Windows.

bakercp, 2015-08-25:
The serial library will compile with using mingw and msys.  Not sure if the included makefile will work out of the box though.  @ElectricRCAircraftGuy you might try installing MSYS2 (for example) and try the makefile in a proper bash shell on windows.  Alternatively I have gotten it to compile in the past with a (slightly out of date, but hopefully still working) toolchain I put together using mingw here:  https://github.com/bakercp/ofmingw  The readme should get you started.

ElectricRCAircraftGuy, 2015-08-25:
thanks @bakercp and @wjwwood . I must admit I'm way in over my head. 

I've installed Microsoft Visual Studio Community 2015 (https://www.visualstudio.com/downloads/download-visual-studio-vs)--it is the replacement for VS Express apparently, and I've built the test program ("test_serial"). It built with no errors, even creating a serial.lib file, which is I think the file I needed to make it run in CodeBlocks. In Visual Studio, however, it still gives me an error when I try to run it.

[image]

I then tried to use CodeBlocks again, this time linking to the .lib file it created (again...no idea what I'm doing). I couldn't get that to work. 

Then I installed win-bash so I could have a windows bash shell. http://win-bash.sourceforge.net/
Running "make" on the library didn't work there either. Same error as before.

Next I installed msys2, per these instructions (http://msys2.github.io/), including typing the commands. It installed. It gives me these three shell options: 

[image]

I don't know what they are. I tried each one of them but don't seem to have access to my local drive with them. It seems to be a virtual Linux system or something. I thought I could run the make command on the library through one of these shells...I think I'm wrong.

So then I went back to win-bash and tried the make command again. 

Then I went back to CodeBlocks to try it again; then I went to your ofmingw repo (https://github.com/bakercp/ofmingw) and began following the instructions. Then I realized I'm totally lost.

I think it's time I search for another serial library. All I need is to send some serial data to an Arduino. I've used serial ports in MATLAB....but I am so lost right now.

I have >>1000 hrs C and C++ programming experience on Arduino, even at the AVR register level, but the extent of my C and C++ programming on a PC is ~1 hr installing CodeBlocks and making a "hello world" pop up in a console.

Again, I'm just super lost. 

I really wanted a cross-platform C++ serial library, but perhaps I need to look for a windows specific one now.

wjwwood, 2015-08-25:
> In Visual Studio, however, it still gives me an error when I try to run it.

It sounds like Visual Studio actually worked, but you cannot run a `.lib`, so you'll need to right click on the example project and run that.

Otherwise, if you really want to use Code Blocks, then you can use a simple `CMakeLists.txt` like this one:

https://github.com/wjwwood/serial/issues/52#issuecomment-28848337

To build my library. You'll have to modify it to build you library, and it might be useful to read up on CMake:

http://www.cmake.org/cmake-tutorial/

Then I believe you can use CMake to generate a CodeBlocks project if that's what you'd like to use:

http://www.cmake.org/cmake/help/v3.0/generator/CodeBlocks.html

So you can follow these steps:

- Clone my repository.
- Replace the CMakeLists.txt with the one I linked above.
- Run the CMake GUI (install it if you don't have it yet).
 - Point the CMake GUI to where you cloned my repository.
 - Point it to a place to generate files (usually a folder you've created called `build`)
 - Select the code blocks generator.
 - Click "configure" and then "generate"
- Open Code Blocks and use the project files put in the target `build` folder you pointed CMake to.

If you're still having trouble, I guess you can look for another library, maybe one with a Windows installer. But any one that has you build it first will probably use Visual Studio or CMake to do so.

ElectricRCAircraftGuy, 2015-08-25:
Ok I'm still struggling through this.....

I discovered the following from your link:

[image]

http://www.cmake.org/cmake/help/v3.0/generator/MinGW%20Makefiles.html#generator:MinGW Makefiles

ElectricRCAircraftGuy, 2015-08-25:
...also I still can't get your visual studio solution to work. And since Visual Studio is so overwhelming I'm trying to stick with CodeBlocks. It seems to be much simpler and ~500x lighter.

wjwwood, 2015-08-25:
I don't use Visual Studio either; the Visual Studio solution was contributed by someone else.

> also I still can't get your visual studio solution to work.

Unfortunately I can't help you with just that as feedback.

I don't know your project requirements, but you could use Code Blocks on Linux. I know for a fact that cmake + Code Blocks works on Linux.

Either way, it sounds like you need to brush up on using CMake on Windows, either with Code Blocks or with the MingGW generator that you linked to. Since Windows isn't my platform of choice, I probably can't help you anymore than I have. Maybe someone else can give you more pointers.

robaato, 2015-09-06:
@ElectricRCAircraftGuy  You cannot use C++ library from MSVC in MinGW project as name mangling schemes are not compatible between MSVC and GCC compiler. It works only for C code.
I just compiled with full success the library and test project using MSVC2013 as this is my usual tool for now and it works perfectly.
To run test project you have to select it on the Solution Explorer window, then use right key and use Set As Startup Project to select it as an active target for debug. Then it will work ... sort of.
As this is a console application which requires parametrs on command line it will start, shows a window for a fraction of a second and this is it.
You can add parameters using: Debug->test_Serial Properties->Configuration Properties->Debugging->Command Arguments.
But it will close the window as soon as an apllication stops. If you want to see the results either add a code to keep the app running for example by changing the main into something like this:
<pre>
int main(int argc, char **argv) {
  int res = 1;
  try {
    res = run(argc, argv);
  } catch (exception &e) {
    cerr << "Unhandled Exception: " << e.what() << endl;
  }
  std::cout << "Press ENTER to finish" << std::endl;
  std::string g;
  std::getline(std::cin, g);
  return res;
}
</pre>
or by running the app from command prompt.

wjwwood, 2015-12-15:
I don't have anymore input on this ticket and it hasn't had any updates lately. I'm going to close it for now.

closed 2015-12-15 by wjwwood

ElectricRCAircraftGuy, 2015-12-15:
Hi all thanks for the responses. I haven't been able to get back to this for a couple months, And still can't quite yet, due to moving. I'll try your suggestions when able.