wjwwood/serial #109: Serial library installation failed

htr99, 2015-08-25:
When I run "sudo make" for installing serial library I get following messages with error.

Can you please tell me what has gone wrong? I'm stuck here for two days now.

cd build && cmake -DCMAKE_INSTALL_PREFIX=/opt/ros/jade/ ..
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/odroid/Downloads/serial-master/build/devel
-- Using CMAKE_PREFIX_PATH: 
-- Found PythonInterp: /usr/bin/python (found version "2.7.6") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/odroid/Downloads/serial-master/build/test_results
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.14
Traceback (most recent call last):
  File "/home/odroid/Downloads/serial-master/build/catkin_generated/generate_cached_setup.py", line 20, in <module>
    from catkin.environment_cache import generate_environment_script
ImportError: No module named catkin.environment_cache
CMake Error at /opt/ros/jade/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/bin/python
  "/home/odroid/Downloads/serial-master/build/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/jade/share/catkin/cmake/all.cmake:186 (safe_execute_process)
  /opt/ros/jade/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/odroid/Downloads/serial-master/build/CMakeFiles/CMakeOutput.log".
See also "/home/odroid/Downloads/serial-master/build/CMakeFiles/CMakeError.log".
make: *** [serial] Error 1

wjwwood, 2015-08-25:
Have you tried deleting your build folder? It looks like the generated code might be out of sync version wise with the stuff in `/opt/ros/jade`? Also make sure you have the latest version of `catkin_pkg`. You can check the version with `python -c "import catkin_pkg; print catkin_pkg.__version__"`

htr99, 2015-08-26:
I tried with deleting build folder with several times. and "python -c "import catkin_pkg; print catkin_pkg.__version__" gives me 0.2.10 as answer , but result is the same. Any other suggestions please.

wjwwood, 2015-08-26:
You can try a `sudo apt-get update && sudo apt-get dist-upgrade` (warning that will update all out of date packages on your machine). It maybe that `catkin_pkg` is too new for what you have, but that seems unlikely to me.

Also you should try just `make` and then `sudo make install`. Perhaps `sudo` is affecting your `PYTHON_PATH`. If you *must* use `sudo` then do `sudo bash` and then `make install`.

htr99, 2015-08-26:
I executed "sudo apt-get update && sudo apt-get dist-upgrade" but nothing is changed. Same output message as previous. I tried without sudo  as well.
Next ..
when I check echo $PYTHONPATH it returned nothing.
Then I changed it using export PYTHONPATH=:/usr/lib/python2.7/dist-packages/catkin_pkg:/opt/ros/jade/lib/python2.7/dist-packages/ .
But when that terminal is closed and run echo $PYTHONPATH , it again return nothing.

So can you please tell me what has gone wrong?

wjwwood, 2015-08-27:
I'm out of ideas, you can try asking on answers.ros.org, since this is actually a problem with catkin not the serial library.

You can also just drop in a simplified CMakeList.txt depending on what you're doing:

https://github.com/wjwwood/serial/issues/52#issuecomment-28848337

jagprieto, 2015-08-28:
Hi...

I have the same problem (Could not find a package configuration file provided by "catkin" ...) on a Xubuntu machine with catkin tools installed. Any idea?

wjwwood, 2015-08-31:
I'm able to build it using the latest version of `catkin` from Jade and the latest version of `catkin_pkg` (I didn't try older versions):

```
william@farl /tmp
% mkdir ws

william@farl /tmp
% cd ws

william@farl /tmp/ws
% mkdir src

william@farl /tmp/ws
% cd src

william@farl /tmp/ws/src
% git clone https://github.com/wjwwood/serial.git
Cloning into 'serial'...
remote: Counting objects: 2542, done.
remote: Total 2542 (delta 0), reused 0 (delta 0), pack-reused 2542
Receiving objects: 100% (2542/2542), 1.50 MiB | 1.45 MiB/s, done.
Resolving deltas: 100% (1189/1189), done.
Checking connectivity... done.

william@farl /tmp/ws/src
% cd ..

william@farl /tmp/ws
% source ~/jade/install/setup.zsh

william@farl /tmp/ws
% catkin_make
Base path: /tmp/ws
Source space: /tmp/ws/src
Build space: /tmp/ws/build
Devel space: /tmp/ws/devel
Install space: /tmp/ws/install
Creating symlink "/tmp/ws/src/CMakeLists.txt" pointing to "/Users/william/jade/install/share/catkin/cmake/toplevel.cmake"
####
#### Running command: "cmake /tmp/ws/src -DCATKIN_DEVEL_PREFIX=/tmp/ws/devel -DCMAKE_INSTALL_PREFIX=/tmp/ws/install -G Unix Makefiles" in "/tmp/ws/build"
####
-- The C compiler identification is AppleClang 6.1.0.6020053
-- The CXX compiler identification is AppleClang 6.1.0.6020053
-- Check for working C compiler: /usr/local/opt/ccache/libexec/cc
-- Check for working C compiler: /usr/local/opt/ccache/libexec/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/local/opt/ccache/libexec/c++
-- Check for working CXX compiler: /usr/local/opt/ccache/libexec/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /tmp/ws/devel
-- Using CMAKE_PREFIX_PATH: /Users/william/jade/install
-- This workspace overlays: /Users/william/jade/install
-- Found PythonInterp: /usr/bin/python (found version "2.7.6")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using default Python package layout
-- Found PY_em: /Library/Python/2.7/site-packages/em.pyc
-- Using empy: /Library/Python/2.7/site-packages/em.pyc
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /tmp/ws/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /usr/local/bin/nosetests-2.7
-- catkin 0.6.14
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 1 packages in topological order:
-- ~~  - serial
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'serial'
-- ==> add_subdirectory(serial)
-- Configuring done
CMake Warning (dev):
  Policy CMP0042 is not set: MACOSX_RPATH is enabled by default.  Run "cmake
  --help-policy CMP0042" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  MACOSX_RPATH is not specified for the following targets:

   serial

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /tmp/ws/build
####
#### Running command: "make -j4 -l4" in "/tmp/ws/build"
####
Scanning dependencies of target serial
[ 50%] [ 50%] [ 75%] Building CXX object serial/CMakeFiles/serial.dir/src/impl/unix.cc.o
Building CXX object serial/CMakeFiles/serial.dir/src/serial.cc.o
Building CXX object serial/CMakeFiles/serial.dir/src/impl/list_ports/list_ports_osx.cc.o
Linking CXX shared library /tmp/ws/devel/lib/libserial.dylib
[ 75%] Built target serial
Scanning dependencies of target serial_example
[100%] Building CXX object serial/CMakeFiles/serial_example.dir/examples/serial_example.cc.o
Linking CXX executable /tmp/ws/devel/lib/serial/serial_example
[100%] Built target serial_example
```

I'll need more information to help.

htr99, 2015-09-02:
Hi wjwwoo,
After running the code mentioned above I did "make" and "sudo make install"  . Then it installed to /tmp/usr/local/ . 

By changing "CMAKE_FLAGS := -DCMAKE_INSTALL_PREFIX=/usr" I could install it to /usr/ directory.

But the problem is when I try example code , "g++ serial_example.cc " it gives following errors, please help me on this.

/tmp/ccJFxeFA.o: In function `enumerate_ports()':
serial_example.cc:(.text+0x2c): undefined reference to `serial::list_ports()'
/tmp/ccJFxeFA.o: In function `run(int, char**)':
serial_example.cc:(.text+0x200): undefined reference to `serial::Serial::Serial(std::string const&, unsigned int, serial::Timeout, serial::bytesize_t, serial::parity_t, serial::stopbits_t, serial::flowcontrol_t)'
serial_example.cc:(.text+0x21e): undefined reference to `serial::Serial::isOpen() const'
serial_example.cc:(.text+0x2de): undefined reference to `serial::Serial::write(std::string const&)'
serial_example.cc:(.text+0x302): undefined reference to `serial::Serial::read(unsigned int)'
serial_example.cc:(.text+0x404): undefined reference to `serial::Serial::write(std::string const&)'
serial_example.cc:(.text+0x428): undefined reference to `serial::Serial::read(unsigned int)'
serial_example.cc:(.text+0x50c): undefined reference to `serial::Serial::write(std::string const&)'
serial_example.cc:(.text+0x52e): undefined reference to `serial::Serial::read(unsigned int)'
serial_example.cc:(.text+0x612): undefined reference to `serial::Serial::write(std::string const&)'
serial_example.cc:(.text+0x636): undefined reference to `serial::Serial::read(unsigned int)'
serial_example.cc:(.text+0x6f6): undefined reference to `serial::Serial::~Serial()'
serial_example.cc:(.text+0x758): undefined reference to `serial::Serial::~Serial()'
/tmp/ccJFxeFA.o: In function `serial::Serial::setTimeout(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int)':
serial_example.cc:(.text._ZN6serial6Serial10setTimeoutEjjjjj[_ZN6serial6Serial10setTimeoutEjjjjj]+0x2e): undefined reference to `serial::Serial::setTimeout(serial::Timeout&)'
collect2: error: ld returned 1 exit status

wjwwood, 2015-09-02:
This is sort of a generic C++ question, maybe it would be more appropriate to ask on StackOverflow.

That being said, you need to link it against the serial shared library.

wjwwood, 2015-12-15:
I'm going to close this for now, please feel free to continue discussion if you like.

closed 2015-12-15 by wjwwood