wjwwood/serial #115: Install problems on Manjaro (~Arch Linux)

korken89, 2015-11-07:
Hi, the install path seems to be wrong for Manjaro (and arch).
`-DCMAKE_INSTALL_PREFIX=/tmp/usr/local` should be `-DCMAKE_INSTALL_PREFIX=/usr/local` if I am not mistaken?
Installing to /tmp will remove it at reboot.

Thank you for your help!

Best Regards
Emil

wjwwood, 2015-11-09:
What documentation or code are you referencing?

I probably instructed users to use `/tmp/usr/local` because the default of `/usr/local` requires sudo. You should only use that if you intend to install it permanently.

korken89, 2015-11-09:
I was trying the (naive) installation outlined in your README for installing the "**Serial Communication Library**".
Indeed to install in `/usr/local` this needs `sudo`, but to install in `/tmp/usr/local` does not seems quite right as after reboot it will be gone.

The 2 ways I would use this (and these are my personal opinions) are:
1. Do a true install and place it in `/use/local/...`, indeed using `sudo`
2. Do a project dependency and include. For example to compile my project with a dependency on `serial` with the code local to the project (for example in catkin)

Would it be possible to add a simple way to choose the install location without modifying the Makefile?
Indeed it is just changing the first line, but still, better if it is officially supported.

What would be your take on this?

Best Regards
Emil

wjwwood, 2015-12-02:
You can setup an install destination env variable which can be set when calling the Makefile to override the install directory using the `?=` make syntax, see: http://stackoverflow.com/questions/448910/makefile-variable-assignment I'd take a pr with this change to the Makefile, but I don't have time to do it myself. I'll close this for now.

closed 2015-12-02 by wjwwood