wjwwood/serial #117: Strange data coming from serial port

karimcitoh, 2015-11-25:
Sorry I am a bit new into serial communication data.

I am running the example provided and the I am getting only weird characters, but I am sure the baud rate I am using is correct. Is there any way to decode it?

Thanks in advance,

[image]

wjwwood, 2015-11-26:
What do you have connected to the serial port? Also please do not post a screen shot, instead copy and paste the text into the issue.

karimcitoh, 2015-11-26:
My excuses for the wrong issue, and also for copying the screenshot. It was my fault, I was not using correct settings in the device I was connecting to!

But thanks for the help and the quick answer.

closed 2015-11-26 by karimcitoh