wjwwood/serial #119: "make" failed.

Matt1989x, 2015-12-15:
Hello,

I'm using Ubuntu 14.04.  I have installed Ros.  I have run "git clone git://github.com/wjwwood/serial.git" and now have the serial folder.  cd into the serial folder.  "make" produces this error.

"cd build && cmake -DCMAKE_INSTALL_PREFIX=/tmp/usr/local ..
CMake Error at CMakeLists.txt:5 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/mattpamenter/serial/build/CMakeFiles/CMakeOutput.log".
make: *** [serial] Error 1"

Does anyone know of a solution? 

I'd really appreciate any help!

Matt

wjwwood, 2015-12-15:
You probably didn't source the ROS setup file. Either that or you don't have `catkin` installed.

Matt1989x, 2015-12-15:
I double check catkin by entering "sudo apt-get install ros-jade-catkin" which gave back: 

Reading package lists... Done
Building dependency tree       
Reading state information... Done
ros-jade-catkin is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.

I'm not sure how to source the ROS setup file. (sorry I'm really new to all of this).  Could you tell me how to do that?

Thanks

wjwwood, 2015-12-15:
`source /opt/ros/jade/setup.bash`

I'd recommend reading up on ROS: http://wiki.ros.org/ROS/StartGuide

Or use serial without `catkin`: https://github.com/wjwwood/serial/issues/52#issuecomment-28848337

If this solves the issue, then please close the ticket. Thanks.

wjwwood, 2016-01-25:
Closing, comment if there is still an issue.

closed 2016-01-25 by wjwwood