wjwwood/serial #120: Trouble connecting at 500kbps

mikepurvis, 2015-12-17:
I have limited access to the hardware producing this stream, so I need to do some more experimenting, but my experience so far is that connecting fails (on both Linux and OS X) when I specify a baud of 500000.

It appears that there's [some intent to support non-standard baud rates](https://github.com/wjwwood/serial/blob/master/src/impl/unix.cc#L303), but I don't know when this was last tested. :|

wjwwood, 2015-12-17:
Typically this doesn't have anything to do with this library, but rather support for the baudrate in the serial hardware and driver. For example, when testing this I have success with some USB-serial converters but not others. I have a encoder that hooked up at 500000 baud in college (when I wrote this), and I remember it working, but I think I was using the serial port on a motherboard and not a converter.

mikepurvis, 2015-12-17:
Upon further investigation, it looks like it is actually working— I was getting an error related to something in my program. Sorry for the noise.

closed 2015-12-17 by mikepurvis