wjwwood/serial #121: Cannot write data

Obbart, 2015-12-28:
Hi, I'm trying to send data between two VSPs created with:
socat -d -d pty,raw,echo=0 pty,raw,echo=0
Opening one port in my program using your library and the other port in putty I cannot send data from program to putty but only read strings from putty.
write returns always zero

wjwwood, 2015-12-28:
I've not used `socat` before. I'd recommend trying your setup with two instances of pyserial or miniterm or something that is "known" to work.

Obbart, 2015-12-29:
I've just tried the communication between two instances of putty and it works just fine in both directions.
It's non mandatory using socat, if you know another method to create a pair of virtual serial ports let me know, i'm just trying to simulate a connection between my software and a robot controller that in this case is a "simulator" running on this pc.
thaks :)

Obbart, 2016-01-02:
in file unix.cc, line 551-557 

while (bytes_read < size) {
    int64_t timeout_remaining_ms = total_timeout.remaining();
    if (timeout_remaining_ms <= 0) {
      // Timed out
      break;
    }
...
}
timeout_remaining_ms is always a negative number so the function deos not write to the port

wjwwood, 2016-01-03:
Make sure you've set the read timeout: https://github.com/wjwwood/serial/blob/1.2.1/include/serial/serial.h#L424-L459

wjwwood, 2016-01-25:
I'm gonna close this. Comment if new information comes up.

closed 2016-01-25 by wjwwood

airduke, 2017-02-17:
I had a similar issue. Problem was actually on the other end of the Serial line not reading properly and the pselect timing out is the symptom I saw from this end.
If you are in a similar spot, try putting a loopback on the other end of the serial as described in the preamble of serial_example.cc to quickly test if you have a similar issue.