wjwwood/serial #126: Unable to compile example on OS X 10.10

dean-shaff, 2016-03-06:
Perhaps this is not the right place to put this, but here goes. I've having issues compiling the code in the example folder. I'm running OS X 10.10. My g++ version is 'Apple LLVM version 6.1.0'. When I try to compile the example code I get the following output:

```
Undefined symbols for architecture x86_64:
  "serial::list_ports()", referenced from:
      enumerate_ports() in serial_example-38ccb9.o
  "serial::Serial::setTimeout(serial::Timeout&)", referenced from:
      serial::Serial::setTimeout(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int) in serial_example-38ccb9.o
  "serial::Serial::read(unsigned long)", referenced from:
      run(int, char**) in serial_example-38ccb9.o
  "serial::Serial::write(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      run(int, char**) in serial_example-38ccb9.o
  "serial::Serial::Serial(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int, serial::Timeout, serial::bytesize_t, serial::parity_t, serial::stopbits_t, serial::flowcontrol_t)", referenced from:
      run(int, char**) in serial_example-38ccb9.o
  "serial::Serial::~Serial()", referenced from:
      run(int, char**) in serial_example-38ccb9.o
  "serial::Serial::isOpen() const", referenced from:
      run(int, char**) in serial_example-38ccb9.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

I thought I would try to install with the `-stdlib=libstdc++` flag, but I get the following error: 

```
Undefined symbols for architecture x86_64:
  "serial::list_ports()", referenced from:
      enumerate_ports() in serial_example-7cc2a4.o
  "serial::Serial::setTimeout(serial::Timeout&)", referenced from:
      serial::Serial::setTimeout(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int) in serial_example-7cc2a4.o
  "serial::Serial::read(unsigned long)", referenced from:
      run(int, char**) in serial_example-7cc2a4.o
  "serial::Serial::write(std::string const&)", referenced from:
      run(int, char**) in serial_example-7cc2a4.o
  "serial::Serial::Serial(std::string const&, unsigned int, serial::Timeout, serial::bytesize_t, serial::parity_t, serial::stopbits_t, serial::flowcontrol_t)", referenced from:
      run(int, char**) in serial_example-7cc2a4.o
  "serial::Serial::~Serial()", referenced from:
      run(int, char**) in serial_example-7cc2a4.o
  "serial::Serial::isOpen() const", referenced from:
      run(int, char**) in serial_example-7cc2a4.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
What am I doing wrong?

wjwwood, 2016-03-07:
Based on `"serial::list_ports()", referenced from: enumerate_ports() in serial_example-38ccb9.o`, I'd say that you are linking against an out of date version of `libserial.dylib`. Make sure you don't have an old copy of serial built or installed some where. Check `/usr/local/lib`, for example. You shouldn't have to change anything for building on OS X, I can do it locally without any issue:

```
william@farl ~/dev/serial/build (git:master:2d416f1)
% make
Scanning dependencies of target serial
[ 16%] Building CXX object CMakeFiles/serial.dir/src/serial.cc.o
[ 33%] Building CXX object CMakeFiles/serial.dir/src/impl/unix.cc.o
[ 50%] Building CXX object CMakeFiles/serial.dir/src/impl/list_ports/list_ports_osx.cc.o
[ 66%] Linking CXX shared library devel/lib/libserial.dylib
[ 66%] Built target serial
Scanning dependencies of target serial_example
[ 83%] Building CXX object CMakeFiles/serial_example.dir/examples/serial_example.cc.o
[100%] Linking CXX executable devel/lib/serial/serial_example
[100%] Built target serial_example

william@farl ~/dev/serial/build (git:master:2d416f1)
% sw_vers
ProductName:	Mac OS X
ProductVersion:	10.11.2
BuildVersion:	15C50
```

I have built it in 10.10 in the past.

wjwwood, 2016-03-07:
Also, if you find a solution, please close the ticket :smile:

closed 2016-03-18 by wjwwood

WeiGuoZoom, 2017-08-04:
Hi, I am also having this issue. 
I am using the cmakelist you mentioned #52 , but when I type "make". It shows the error: 
Scanning dependencies of target serial
[ 20%] Building CXX object CMakeFiles/serial.dir/src/serial.cc.o
[ 40%] Building CXX object CMakeFiles/serial.dir/src/impl/unix.cc.o
[ 60%] Linking CXX static library libserial.a
[ 60%] Built target serial
Scanning dependencies of target serial_example
[ 80%] Building CXX object CMakeFiles/serial_example.dir/examples/serial_example.cc.o
[100%] Linking CXX executable serial_example
Undefined symbols for architecture x86_64:
  "serial::list_ports()", referenced from:
      enumerate_ports() in serial_example.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[3]: *** [serial_example] Error 1
make[2]: *** [CMakeFiles/serial_example.dir/all] Error 2
make[1]: *** [all] Error 2
make: *** [serial] Error 2

please kindly advice.