wjwwood/serial #128: How to add the library to my code after "make install"

dottantgal, 2016-05-23:
Hi, I wanna add the library to my own code, but I don't know how to include it in the Cmakelists file.
It gives me an error  "fatal error: serial/serial.h"
This is how my CMakeLists looks like
`cmake_minimum_required(VERSION 2.8.3)
project(serial_sensor)

# Set the name to use for the executable.
set (BINNAME serial_sensor)
## Set the source files to use with the executable.
set (SRCS1 ${SRCS1} src/serial_sensor.cpp)
## Find catkin macros and libraries
## if COMPONENTS list like find_package(catkin REQUIRED COMPONENTS xyz)
## is used, also find other catkin packages
find_package(catkin REQUIRED COMPONENTS
  roscpp
  rospy
  std_msgs
  tf
  message_generation
)

## Generate messages in the 'msg' folder
add_message_files(
  FILES
  Altitude.msg
  Heading.msg
  Pressure.msg
)

## Generate added messages and services with any dependencies listed here
generate_messages(
   DEPENDENCIES
   std_msgs
)

catkin_package(
#  INCLUDE_DIRS include
  LIBRARIES serial
  CATKIN_DEPENDS message_runtime
#  CATKIN_DEPENDS roscpp rospy std_msgs
#  DEPENDS system_lib
)

include_directories(${CMAKE_CURRENT_SOURCE_DIR})
include_directories(include ${catkin_INCLUDE_DIRS})

add_executable(serial_motors src/serial_motors.cpp)
target_link_libraries(serial_motors ${catkin_LIBRARIES})

add_executable (${BINNAME} ${SRCS1})
add_dependencies(${BINNAME} ${PROJECT_NAME}_generate_messages_cpp)
target_link_libraries(${BINNAME} ${catkin_LIBRARIES})
`
I ve followed the "how to" till the "make install" and this last command gives me
`Install the project...
-- Install configuration: ""
-- Installing: /tmp/usr/local/_setup_util.py
-- Installing: /tmp/usr/local/env.sh
-- Installing: /tmp/usr/local/setup.bash
-- Installing: /tmp/usr/local/setup.sh
-- Installing: /tmp/usr/local/setup.zsh
-- Installing: /tmp/usr/local/.rosinstall
-- Installing: /tmp/usr/local/lib/pkgconfig/serial.pc
-- Installing: /tmp/usr/local/share/serial/cmake/serialConfig.cmake
-- Installing: /tmp/usr/local/share/serial/cmake/serialConfig-version.cmake
-- Installing: /tmp/usr/local/share/serial/package.xml
-- Installing: /tmp/usr/local/lib/libserial.so
-- Installing: /tmp/usr/local/include/serial/serial.h
-- Installing: /tmp/usr/local/include/serial/v8stdint.h`
is it strange that it installs under the tmp folder?
Thank you

wjwwood, 2016-05-23:
You can set the installation directory using the `CMAKE_INSTALL_PREFIX`. If you're using ROS then your `CMakeLists.txt` above looks ok. If you're using ROS and Ubuntu, you can just install my package as `sudo apt-get install ros-indigo-serial` (you can replace `indigo` with whatever ROS distro you are using). There is also an issue where I describe how you can integrate it into your own CMake project: https://github.com/wjwwood/serial/issues/52#issuecomment-28848337

LivioBrunner4, 2019-12-03:
According to https://packages.ubuntu.com/ the mentioned package: "ros-indigo-serial" (as well as ros-melodic-serial) does not exist.
Is it only me, who does not find the package?

romainreignier, 2019-12-12:
Le packages with the prefix `ros-<ros_distro>-` are not available in the official Ubuntu repositories but in the ROS repositories that you must manually setup according to this page: https://wiki.ros.org/melodic/Installation/Ubuntu
You only need the 1.1, 1.2 and 1.3 steps.