wjwwood/serial #134: serial Unable to open port.

j7451229, 2016-07-26:
HI everyone!

I used wjwwood_serial to connect my RS232.And i installed by use sudo apt-get install ros-indigo-serial in computer1 and laptop.

Then I created package and wrote code, let it can listen to ROS node. And it can run perfect in computer1,but when i copy this package to laptop and run. It shows Unable to open port.

my code:https://drive.google.com/folderview?id=0B02FisAlf-IXTGVmclV0UW11QTg&usp=sharing

what happened? thank for any help and advice!

best regard!

wjwwood, 2016-07-26:
There is no source code in the Google drive and you have not included the output of the program. We cannot help you with this level of information. I'll reopen the issue when/if you update the information.

closed 2016-07-26 by wjwwood

j7451229, 2016-07-27:
I sloved the problem by use sudo chmod 666 /dev/ttyUSB0. it is the permission problem!