wjwwood/serial #139: Custom Baudrate OSX

saynono, 2016-09-28:
Hi. 
It seems that following line:441 in unix Serial::SerialImpl::reconfigurePort resets custom baudrates on OSX.
```
  // activate settings
  ::tcsetattr (fd_, TCSANOW, &options);
```

Setting custom baudrates after this line fixes this as far as I can tell.
Also for noobs as myself, might be helpfull mentioning to call this on OSX:

`stty ospeed <custom_baudrate>`
`stty ispeed <custom_baudrate>
`

and on linux

`sudo stty -F <port> <custom_baudrate>
`

My changes here:
https://github.com/saynono/serial/commit/c997027b3969d595d9c4c8aa2dda9fe3f613e688

willemkempers, 2017-06-19:
I had the same issue, but found adding the following solution to line 343 a bit more elegant:

```
#if defined(MAC_OS_X_VERSION_10_4) && (MAC_OS_X_VERSION_MIN_REQUIRED >= MAC_OS_X_VERSION_10_4)
  else {
      speed_t new_baud = static_cast<speed_t> (baudrate_);
      ::cfsetispeed(&options, new_baud);
      ::cfsetospeed(&options, new_baud);
  }
#endif
```

Tested this super quickly on my macbook running macOS 10.12

atomiechen, 2020-03-21:
> Setting custom baudrates after this line fixes this as far as I can tell.

@saynono This is correct as far as I can see. I look into source code of PySerial (in `serialposix.py`):

```python
def _reconfigure_port(self, force_update=False):
    ...
    # activate settings
    if force_update or [iflag, oflag, cflag, lflag, ispeed, ospeed, cc] != orig_attr:
        termios.tcsetattr(
            self.fd,
            termios.TCSANOW,
            [iflag, oflag, cflag, lflag, ispeed, ospeed, cc])

    # apply custom baud rate, if any
    if custom_baud is not None:
        self._set_special_baudrate(custom_baud)
```

Since I can use PySerial to connect my Arduino by 250000 baudrate yet I cannot do this via this c++ serial library, I believe this is at least a big difference I can find. Same issue also showed up in other project and a similar solution here: [Custom baud rate in OSX won't work #57, npat-efault/picocom](https://github.com/npat-efault/picocom/issues/57).

In my scenario the problem is solved on macOS Mojave 10.14.4. My commit: 47732659f5e17c96b0414c513979131202073083