wjwwood/serial #144: cable break

vasvladal, 2016-11-14:
How do you know that he was taken out cable. I take out the cable, but isOpen shows that everything is OK

dontsovcmc, 2016-11-22:
COM port can't understand cable state. You can only see to "timeout" option and tune it for your device. If cable break - no answer, but COM port is open.
It is not library problem.

wjwwood, 2016-11-28:
As @dontsovcmc said, the state of the cable's connection is not exposed by the OS, so the library cannot know when it is unplugged. A USB-serial adapter being unplugged is different, because the virtual COM port is often removed by the OS, which the library does detect.

closed 2016-11-28 by wjwwood

vasvladal, 2016-12-07:
Thanks for the explanation. Another question. After disconnecting the cables when trying to write to the port there is an exception. How to handle this situation?

vasvladal, 2016-12-07:
The question was solved by compiler setting Enabled C++ Exceptions  to Yes with SEH Exceptions (/ EHa)