wjwwood/serial #151: build error while declare a serial project in private of ROS class

CrazyPopLin, 2017-04-21:
First I test the serial package in a simple ROS publisher to get serial data, and it works fine. the code shown below: 
**simple code which is right**
```
#include "ros/ros.h"
#include "std_msgs/String.h"
#include "serial/serial.h"
#include <sstream>
#include <string>

using namespace std;

int main(int argc, char **argv)
{
  ros::init(argc,argv, "talker");
  ros::NodeHandle n;
  ros::Publisher test_pub=n.advertise<std_msgs::String>("chatter",1000);

  ros::Rate loop_rate(10);

  string port = "/dev/Lidar_Scanner";
  unsigned long baud = 115200;

  serial::Serial my_serial(port, baud, serial::Timeout::simpleTimeout(250));

  int count = 0;
  while (ros::ok())
  {
	//deal with incoming serial data
    uint8_t *buffer_ = new uint8_t[6];
    my_serial.read(buffer_,6);

	char start = buffer_[0];
	int a1= buffer_[1];
	int a2= buffer_[2];
	int a3= buffer_[3];
	int a4= buffer_[4];
	char end = buffer_[5];

    delete[] buffer_;

	int range_, angle_;
	if(start == '#' && end == '>' ){
  		angle_ = (a1 <<8) | (a2 & 0xff);
		range_ = (a3 <<8) | (a4 & 0xff);

		cout<< ", angle: " << angle_;
		cout<< ", range: " << range_ <<endl;
	}
	else if (start =='*' && end == '>'){

		angle_ = (a1 <<8) | (a2 & 0xff);
		range_ = (a3 <<8) | (a4 & 0xff);

		cout<< ", angle: " << angle_;
		cout<< ", range: " << range_ <<endl;
	}
	else{
		count++;
		cout << "this is No. "<<count << "error!"<<endl;
	}

	//ROS publihser
	std_msgs::String msg;
        msg.data = "a";
	test_pub.publish(msg);
        ros::spinOnce();

    //loop_rate.sleep();
  }

  return 0;
}

```
Then I declare serial project in a class as private part for future complex implement, I got build error. I just use same declare information the one used in first test. and it's works. Any suggestion will be appreciated. The code and errors shown below: 
**test code which is wrong**
```
#include "ros/ros.h"
#include "std_msgs/String.h"
#include <sensor_msgs/LaserScan.h>
#include <tf/transform_broadcaster.h>
#include "serial/serial.h"

#include <sstream>
#include <string>
#include <map>

using namespace std;

class LidarScanner_Driver
{
	public:
		LidarScanner_Driver()
		{
			scan_pub = nh.advertise<sensor_msgs::LaserScan>( "Scan", 50);
		}
		void extractData( int &angle_, int &range_);

	private:
		ros::NodeHandle nh;
		ros::Publisher scan_pub;

		string port = "/dev/ttyUSB0";
		unsigned long baud = 115200;
		serial::Serial my_serial(port, baud, serial::Timeout::simpleTimeout(250));
};
		
void LidarScanner_Driver::extractData(int &angle_, int &range_)
{
	//deal with incoming serial data
        uint8_t *buffer_ = new uint8_t[6];
        my_serial.read(buffer_,6);

	char start = buffer_[0];
	int a1= buffer_[1];
	int a2= buffer_[2];
	int a3= buffer_[3];
	int a4= buffer_[4];
	char end = buffer_[5];
        delete[] buffer_;

	//check if this serial stream is correct which means every byte is received
    if ((start == '#' && end == '>')|| (start == '*' && end == '>')){
		angle_ = (a1 <<8) | (a2 & 0xff);
		range_ = (a3 <<8) | (a4 & 0xff);
	}	
	else{
		angle_= -1;
		range_= -1;
	}
}

int main(int argc, char **argv)
{
  //setup ROS
  ros::init(argc,argv, "lidarscanner_driver");
  LidarScanner_Driver Driver;
  ros::Rate loop_rate(100);

  while(ros::ok())
  {
	int angle, range;
	Driver.extractData( angle, range);

	cout<<"angle: " << angle;
	cout<<"range: " << range<<endl;

	ros::spinOnce();
	//loop_rate.sleep();
  }
  return 0;
}
```  
I got error like this:  
**errors**

> /home/lin/catkin_ws/src/lidar_scanner/src/test1.cpp:33:28: error: ‘port’ is not a type
>    serial::Serial my_serial(port, baud, serial::Timeout::simpleTimeout(250));
>                             ^
> /home/lin/catkin_ws/src/lidar_scanner/src/test1.cpp:33:34: error: ‘baud’ is not a type
>    serial::Serial my_serial(port, baud, serial::Timeout::simpleTimeout(250));
>                                   ^
> /home/lin/catkin_ws/src/lidar_scanner/src/test1.cpp:33:57: error: ‘serial::Timeout::simpleTimeout’ is not a type
>    serial::Serial my_serial(port, baud, serial::Timeout::simpleTimeout(250));
>                                                          ^
> /home/lin/catkin_ws/src/lidar_scanner/src/test1.cpp:33:71: error: expected ‘)’ before numeric constant
>    serial::Serial my_serial(port, baud, serial::Timeout::simpleTimeout(250));
>                                                                        ^
> /home/lin/catkin_ws/src/lidar_scanner/src/test1.cpp:33:71: error: expected ‘)’ before numeric constant
> /home/lin/catkin_ws/src/lidar_scanner/src/test1.cpp:33:70: error: expected ‘;’ at end of member declaration
>    serial::Serial my_serial(port, baud, serial::Timeout::simpleTimeout(250));
>                                                                       ^
> /home/lin/catkin_ws/src/lidar_scanner/src/test1.cpp:33:71: error: expected unqualified-id before numeric constant
>    serial::Serial my_serial(port, baud, serial::Timeout::simpleTimeout(250));

romainreignier, 2017-04-21:
Your issue is not related to this library but your understanding of C++ class member initialization.
Your class should look like this:

```C++
class LidarScanner_Driver
{
	public:
		LidarScanner_Driver()
                : port("/dev/ttyUSB0"), baud(115200), my_serial(port, baud, serial::Timeout::simpleTimeout(250))
		{
			scan_pub = nh.advertise<sensor_msgs::LaserScan>( "Scan", 50);
		}
		void extractData( int &angle_, int &range_);

	private:
		ros::NodeHandle nh;
		ros::Publisher scan_pub;

		string port;
		unsigned long baud;
		serial::Serial my_serial;
};
```

CrazyPopLin, 2017-04-21:
thanks, you are right

closed 2017-04-21 by CrazyPopLin