wjwwood/serial #152: undefined reference to `serial::list_ports()'    2

jomilam, 2017-05-17:
My issue is similar to issue #78, but after telling the linker where to find the library and trying a couple other things, the problem persists. Also, my installation put the headers and the library in a tmp folder instead of the usual include directories (headers in /tmp/usr/local/include and library in /tmp/usr/local/lib as opposed to the usual /usr/local/include and /usr/local/lib respectively). After telling the compiler where to look for both headers and library with the command, 
 `g++ serial_example.cc  -o serial -L/tmp/usr/local/lib/libserial.so -I/tmp/usr/local/include`
 it can find them but still fails to compile. I ran the "make test", results below:

$ make test
cd build && cmake -DCMAKE_INSTALL_PREFIX=/tmp/usr/local ..
-- Using CATKIN_DEVEL_PREFIX: /home/localadmin/apps/serial-master/build/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/indigo
-- This workspace overlays: /opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/localadmin/apps/serial-master/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- Configuring done
-- Generating done
-- Build files have been written to: /home/localadmin/apps/serial-master/build
cd build && make run_tests
make[1]: Entering directory `/home/localadmin/apps/serial-master/build'
make[2]: Entering directory `/home/localadmin/apps/serial-master/build'
make[3]: Entering directory `/home/localadmin/apps/serial-master/build'
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
Scanning dependencies of target gtest
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
[  8%] Building CXX object gtest/CMakeFiles/gtest.dir/src/gtest-all.cc.o
[ 16%] Linking CXX shared library libgtest.so
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
[ 16%] Built target gtest
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
Scanning dependencies of target gtest_main
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
[ 25%] Building CXX object gtest/CMakeFiles/gtest_main.dir/src/gtest_main.cc.o
[ 33%] Linking CXX shared library libgtest_main.so
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
[ 33%] Built target gtest_main
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
[ 66%] Built target serial
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
Scanning dependencies of target serial-test
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
[ 75%] Building CXX object tests/CMakeFiles/serial-test.dir/unix_serial_tests.cc.o
[ 83%] Linking CXX executable ../devel/lib/serial/serial-test
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
[ 83%] Built target serial-test
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
Scanning dependencies of target serial-test-timer
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
[ 91%] Building CXX object tests/CMakeFiles/serial-test-timer.dir/unit/unix_timer_tests.cc.o
[100%] Linking CXX executable ../devel/lib/serial/serial-test-timer
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
[100%] Built target serial-test-timer
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
Scanning dependencies of target tests
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
[100%] Built target tests
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
Scanning dependencies of target clean_test_results_serial
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
Removing test result files from '/home/localadmin/apps/serial-master/build/test_results/serial'
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
[100%] Built target clean_test_results_serial
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
Scanning dependencies of target _run_tests_serial_gtest_serial-test
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
-- run_tests.py: execute commands
  /home/localadmin/apps/serial-master/build/devel/lib/serial/serial-test --gtest_output=xml:/home/localadmin/apps/serial-master/build/test_results/serial/gtest-serial-test.xml
[==========] Running 4 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 4 tests from SerialTests
[ RUN      ] SerialTests.readWorks
[       OK ] SerialTests.readWorks (227 ms)
[ RUN      ] SerialTests.writeWorks
[       OK ] SerialTests.writeWorks (0 ms)
[ RUN      ] SerialTests.timeoutWorks
[       OK ] SerialTests.timeoutWorks (249 ms)
[ RUN      ] SerialTests.partialRead
[       OK ] SerialTests.partialRead (250 ms)
[----------] 4 tests from SerialTests (727 ms total)

[----------] Global test environment tear-down
[==========] 4 tests from 1 test case ran. (728 ms total)
[  PASSED  ] 4 tests.
-- run_tests.py: verify result "/home/localadmin/apps/serial-master/build/test_results/serial/gtest-serial-test.xml"
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
[100%] Built target _run_tests_serial_gtest_serial-test
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
Scanning dependencies of target _run_tests_serial_gtest_serial-test-timer
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
-- run_tests.py: execute commands
  /home/localadmin/apps/serial-master/build/devel/lib/serial/serial-test-timer --gtest_output=xml:/home/localadmin/apps/serial-master/build/test_results/serial/gtest-serial-test-timer.xml
[==========] Running 2 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 2 tests from timer_tests
[ RUN      ] timer_tests.short_intervals
[       OK ] timer_tests.short_intervals (933 ms)
[ RUN      ] timer_tests.overlapping_long_intervals
[       OK ] timer_tests.overlapping_long_intervals (1011 ms)
[----------] 2 tests from timer_tests (1944 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test case ran. (1944 ms total)
[  PASSED  ] 2 tests.
-- run_tests.py: verify result "/home/localadmin/apps/serial-master/build/test_results/serial/gtest-serial-test-timer.xml"
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
[100%] Built target _run_tests_serial_gtest_serial-test-timer
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
Scanning dependencies of target _run_tests_serial_gtest
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
[100%] Built target _run_tests_serial_gtest
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
Scanning dependencies of target _run_tests_serial
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
[100%] Built target _run_tests_serial
make[4]: Entering directory `/home/localadmin/apps/serial-master/build'
Scanning dependencies of target run_tests
make[4]: Leaving directory `/home/localadmin/apps/serial-master/build'
[100%] Built target run_tests
make[3]: Leaving directory `/home/localadmin/apps/serial-master/build'
make[2]: Leaving directory `/home/localadmin/apps/serial-master/build'
make[1]: Leaving directory `/home/localadmin/apps/serial-master/build'

slivingston, 2017-08-07:
@jomilam can you include some of the error message that you receive when you try the command in the opening post?

    g++ serial_example.cc -o serial -L/tmp/usr/local/lib/libserial.so -I/tmp/usr/local/include

jomilam, 2017-08-25:
I re-cmake/make/installed and it seemed to put the library and header in the usual places, so that is not a problem anymore. 
Trying again to compile but with just `g++ serial_example.cc -o serial` the output error is:

/tmp/ccLwdBgg.o: In function `enumerate_ports()':
serial_example.cc:(.text+0x32): undefined reference to `serial::list_ports()'
/tmp/ccLwdBgg.o: In function `run(int, char**)':
serial_example.cc:(.text+0x2a1): undefined reference to `serial::Serial::Serial(std::string const&, unsigned int, serial::Timeout, serial::bytesize_t, serial::parity_t, serial::stopbits_t, serial::flowcontrol_t)'
serial_example.cc:(.text+0x2bc): undefined reference to `serial::Serial::isOpen() const'
serial_example.cc:(.text+0x37d): undefined reference to `serial::Serial::write(std::string const&)'
serial_example.cc:(.text+0x3a4): undefined reference to `serial::Serial::read(unsigned long)'
serial_example.cc:(.text+0x4b7): undefined reference to `serial::Serial::write(std::string const&)'
serial_example.cc:(.text+0x4de): undefined reference to `serial::Serial::read(unsigned long)'
serial_example.cc:(.text+0x5c8): undefined reference to `serial::Serial::write(std::string const&)'
serial_example.cc:(.text+0x5ee): undefined reference to `serial::Serial::read(unsigned long)'
serial_example.cc:(.text+0x6d8): undefined reference to `serial::Serial::write(std::string const&)'
serial_example.cc:(.text+0x6ff): undefined reference to `serial::Serial::read(unsigned long)'
serial_example.cc:(.text+0x7c8): undefined reference to `serial::Serial::~Serial()'
serial_example.cc:(.text+0x85c): undefined reference to `serial::Serial::~Serial()'
/tmp/ccLwdBgg.o: In function `serial::Serial::setTimeout(unsigned int, unsigned int, unsigned int, unsigned int, unsigned int)':
serial_example.cc:(.text._ZN6serial6Serial10setTimeoutEjjjjj[_ZN6serial6Serial10setTimeoutEjjjjj]+0x4e): undefined reference to `serial::Serial::setTimeout(serial::Timeout&)'
collect2: error: ld returned 1 exit status

After running make test again, output is:

`Running tests...
Test project /home/localadmin/apps/serial-master/build
    Start 1: _ctest_serial_gtest_serial-test
1/2 Test #1: _ctest_serial_gtest_serial-test .........***Failed    0.04 sec
    Start 2: _ctest_serial_gtest_serial-test-timer
2/2 Test #2: _ctest_serial_gtest_serial-test-timer ...***Failed    0.04 sec

0% tests passed, 2 tests failed out of 2

Total Test time (real) =   0.09 sec

The following tests FAILED:
	  1 - _ctest_serial_gtest_serial-test (Failed)
	  2 - _ctest_serial_gtest_serial-test-timer (Failed)
Errors while running CTest
make: *** [test] Error 8
`

slivingston, 2017-08-25:
I would not expect the bare `g++ serial_example.cc -o serial` to succeed. At least you need to tell the compiler where to find header files and the linker where to find object files.

Have you tried to make a custom CMakeLists.txt file for your case? If you are building without ROS, then there is one example shown at <https://github.com/wjwwood/serial/issues/52#issuecomment-28848337>. It may require some additions for `list_ports`, as noted at <https://github.com/wjwwood/serial/issues/156#issuecomment-320784535>