wjwwood/serial #156: Issue on complie

WeiGuoZoom, 2017-08-07:
I am using the cmakelist you mentioned #52 , but when I type "make". It shows the error:
cd build && cmake -DCMAKE_INSTALL_PREFIX=/tmp/usr/local .. -- The C compiler identification is AppleClang 8.1.0.8020038 -- The CXX compiler identification is AppleClang 8.1.0.8020038 -- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works -- Detecting C compiler ABI info -- Detecting C compiler ABI info - done -- Detecting C compile features -- Detecting C compile features - done -- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works -- Detecting CXX compiler ABI info -- Detecting CXX compiler ABI info - done -- Detecting CXX compile features -- Detecting CXX compile features - done -- Configuring done -- Generating done -- Build files have been written to: /Users/weiguo/iCloud Drive (Archive)/Documents/zoom/cppRs232/serial/build cd build && /Applications/Xcode.app/Contents/Developer/usr/bin/make Scanning dependencies of target serial [ 20%] Building CXX object CMakeFiles/serial.dir/src/serial.cc.o [ 40%] Building CXX object CMakeFiles/serial.dir/src/impl/unix.cc.o [ 60%] Linking CXX static library libserial.a [ 60%] Built target serial Scanning dependencies of target serial_example [ 80%] Building CXX object CMakeFiles/serial_example.dir/examples/serial_example.cc.o [100%] Linking CXX executable serial_example Undefined symbols for architecture x86_64: "serial::list_ports()", referenced from: enumerate_ports() in serial_example.cc.o ld: symbol(s) not found for architecture x86_64 clang: error: linker command failed with exit code 1 (use -v to see invocation) make[3]: *** [serial_example] Error 1 make[2]: *** [CMakeFiles/serial_example.dir/all] Error 2 make[1]: *** [all] Error 2 make: *** [serial] Error 2

please kindly advice.

slivingston, 2017-08-07:
the example uses `list_ports`, which requires adding an appropriate file from under [src/impl/list_ports](https://github.com/wjwwood/serial/tree/827c4a784dd4fdd35dc391f37ef152eab7c9c9b2/src/impl/list_ports), depending on the host platform.

from your build error message, it appears that you are on OSX, so try adding "src/impl/list_ports/list_ports_osx.cc" to `serial_SRCS` in the CMakeLists.txt

WeiGuoZoom, 2017-08-07:
Hi @slivingston , thanks for replying, 
Now my CMakeLists.txt looks like below:
```
cmake_minimum_required(VERSION 2.8.3)
project(my_project)

include_directories(include)

set(serial_SRCS src/serial.cc include/serial/serial.h include/serial/v8stdint.h src/impl/list_ports/list_ports_osx.cc)
if(UNIX)
    list(APPEND serial_SRCS src/impl/unix.cc)
else()
    list(APPEND serial_SRCS src/impl/win.cc)
endif()

## Add serial library
add_library(serial ${serial_SRCS})
if(UNIX AND NOT APPLE)
    target_link_libraries(serial rt)
endif()

## Build your executable
add_executable(serial_example examples/serial_example.cc)
## Link your executable against the library
target_link_libraries(serial_example serial)
```
but it still throw me error as follow: 
`Scanning dependencies of target serial
[ 16%] Building CXX object CMakeFiles/serial.dir/src/serial.cc.o
[ 33%] Building CXX object CMakeFiles/serial.dir/src/impl/list_ports/list_ports_osx.cc.o
[ 50%] Building CXX object CMakeFiles/serial.dir/src/impl/unix.cc.o
[ 66%] Linking CXX static library libserial.a
[ 66%] Built target serial
[ 83%] Building CXX object CMakeFiles/serial_example.dir/examples/serial_example.cc.o
[100%] Linking CXX executable serial_example
Undefined symbols for architecture x86_64:
  "_CFDictionarySetValue", referenced from:
      serial::list_ports() in libserial.a(list_ports_osx.cc.o)
  "_CFGetTypeID", referenced from:
      get_device_path(unsigned int&) in libserial.a(list_ports_osx.cc.o)
      get_string_property(unsigned int&, char const*) in libserial.a(list_ports_osx.cc.o)
      get_int_property(unsigned int&, char const*) in libserial.a(list_ports_osx.cc.o)
  "_CFNumberGetTypeID", referenced from:
      get_int_property(unsigned int&, char const*) in libserial.a(list_ports_osx.cc.o)
  "_CFNumberGetValue", referenced from:
      get_int_property(unsigned int&, char const*) in libserial.a(list_ports_osx.cc.o)
  "_CFRelease", referenced from:
      get_device_path(unsigned int&) in libserial.a(list_ports_osx.cc.o)
      get_string_property(unsigned int&, char const*) in libserial.a(list_ports_osx.cc.o)
      get_int_property(unsigned int&, char const*) in libserial.a(list_ports_osx.cc.o)
  "_CFStringCreateWithCString", referenced from:
      get_string_property(unsigned int&, char const*) in libserial.a(list_ports_osx.cc.o)
      get_int_property(unsigned int&, char const*) in libserial.a(list_ports_osx.cc.o)
  "_CFStringGetCString", referenced from:
      cfstring_to_string(__CFString const*) in libserial.a(list_ports_osx.cc.o)
  "_CFStringGetTypeID", referenced from:
      get_device_path(unsigned int&) in libserial.a(list_ports_osx.cc.o)
      get_string_property(unsigned int&, char const*) in libserial.a(list_ports_osx.cc.o)
  "_IOIteratorNext", referenced from:
      serial::list_ports() in libserial.a(list_ports_osx.cc.o)
  "_IOMasterPort", referenced from:
      serial::list_ports() in libserial.a(list_ports_osx.cc.o)
  "_IOObjectGetClass", referenced from:
      get_class_name(unsigned int&) in libserial.a(list_ports_osx.cc.o)
  "_IOObjectRelease", referenced from:
      serial::list_ports() in libserial.a(list_ports_osx.cc.o)
  "_IORegistryEntryCreateCFProperty", referenced from:
      get_device_path(unsigned int&) in libserial.a(list_ports_osx.cc.o)
      get_string_property(unsigned int&, char const*) in libserial.a(list_ports_osx.cc.o)
      get_int_property(unsigned int&, char const*) in libserial.a(list_ports_osx.cc.o)
  "_IORegistryEntryGetParentEntry", referenced from:
      get_parent_iousb_device(unsigned int&) in libserial.a(list_ports_osx.cc.o)
  "_IOServiceGetMatchingServices", referenced from:
      serial::list_ports() in libserial.a(list_ports_osx.cc.o)
  "_IOServiceMatching", referenced from:
      serial::list_ports() in libserial.a(list_ports_osx.cc.o)
  "___CFConstantStringClassReference", referenced from:
      CFString in libserial.a(list_ports_osx.cc.o)
      CFString in libserial.a(list_ports_osx.cc.o)
      CFString in libserial.a(list_ports_osx.cc.o)
  "_kCFAllocatorDefault", referenced from:
      get_device_path(unsigned int&) in libserial.a(list_ports_osx.cc.o)
      get_string_property(unsigned int&, char const*) in libserial.a(list_ports_osx.cc.o)
      get_int_property(unsigned int&, char const*) in libserial.a(list_ports_osx.cc.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [serial_example] Error 1
make[1]: *** [CMakeFiles/serial_example.dir/all] Error 2
make: *** [all] Error 2`
what could go wrong here? Thanks

slivingston, 2017-08-07:
I do not currently have access to OSX with which to test, but comparing to [the root CMakeLists.txt](https://github.com/wjwwood/serial/blob/827c4a784dd4fdd35dc391f37ef152eab7c9c9b2/CMakeLists.txt), some additional work is required to compile the `list_ports` method on OSX.

In particular, try additions to CMakeLists.txt as from commit 31a589e6e72f00ae354237d7edd7e7aa7a691242

slivingston, 2017-08-20:
@WeiGuoZoom did you have success?

WeiGuoZoom, 2017-08-21:
Hi, It's working now. Thanks for the help!

slivingston, 2017-08-22:
good. then, can you close this issue?

closed 2017-08-22 by WeiGuoZoom

pzdkn, 2019-11-14:
@WeiGuoZoom , how did you resolve it ?