wjwwood/serial #161: flush() does not work for Windows and Linux

vasek-cz, 2017-10-03:
According to your documentation the `flush()` function should flush both input and output buffers. However both for Windows and Linux platforms it doesn't work properly - buffers are not cleared.

Correct and tested implementation:

Windows
 `PurgeComm(fd_, PURGE_RXCLEAR | PURGE_TXCLEAR);`

Linux
   `tcflush(fd_, TCIOFLUSH);`

sirop, 2017-11-16:
Because of

> If hFile is a handle to a communications device, the function only flushes the transmit buffer.

in case of BOOL WINAPI FlushFileBuffers(  _In_ HANDLE hFile )
and

> The tcdrain() function shall block until all output written to the object referred to by fildes is transmitted.

in Linux case?

sirop, 2017-11-16:
Confiremed on Windows 8.1:
```
// port, baudrate, timeout in milliseconds
  serial::Serial my_serial(port, 38400, serial::Timeout::Timeout(0,1,0,0,0));

  if(my_serial.isOpen()) {
    cout << port << " opened? " << " Yes." << endl;
    my_serial.flush();
    string result = my_serial.read(72);
    if (result.length()>0) 
        std::cout << result.length() << endl;
   }
```
always yields 72, which means that  `my_serial.flush();` does not function as new 72 bytes cannot be delivered within 1 ms read timeout if the flush was successful before.

wjwwood, 2018-01-13:
Thanks for reporting the issue and for trying to confirm it with an example.

It sounds like you have a suggested solution, please open a pull request:

https://help.github.com/articles/creating-a-pull-request/

Thanks!

sirop, 2018-01-14:
@wjwwood 
If I am addressed, please give me a day or two,