wjwwood/serial #162: Compilation Issues on Cygwin

helloguys1913, 2017-11-01:
**I am getting the following error when I do make on Cygwin Compiler.....**

/home/BHANDPX3/serial-lib/src/impl/unix.cc: In member function ‘bool serial::Serial::SerialImpl::waitReadable(uint32_t)’:
/home/BHANDPX3/serial-lib/src/impl/unix.cc:499:68: error: ‘pselect’ was not declared in this scope
   int r = pselect (fd_ + 1, &readfds, NULL, NULL, &timeout_ts, NULL);
                                                                    ^
/home/BHANDPX3/serial-lib/src/impl/unix.cc: In member function ‘void serial::Serial::SerialImpl::waitByteTimes(std::size_t)’:
/home/BHANDPX3/serial-lib/src/impl/unix.cc:526:49: error: ‘pselect’ was not declared in this scope
   pselect (0, NULL, NULL, NULL, &wait_time, NULL);
                                                 ^
/home/BHANDPX3/serial-lib/src/impl/unix.cc: In member function ‘size_t serial::Serial::SerialImpl::write(const uint8_t*, std::size_t)’:
/home/BHANDPX3/serial-lib/src/impl/unix.cc:637:68: error: ‘pselect’ was not declared in this scope
     int r = pselect (fd_ + 1, NULL, &writefds, NULL, &timeout, NULL);
                                                                    ^
/home/BHANDPX3/serial-lib/src/impl/unix.cc: In member function ‘bool serial::Serial::SerialImpl::waitForChange()’:
/home/BHANDPX3/serial-lib/src/impl/unix.cc:926:16: error: ‘usleep’ was not declared in this scope
     usleep(1000);
                ^
make[2]: *** [CMakeFiles/serial.dir/build.make:87: CMakeFiles/serial.dir/src/impl/unix.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:106: CMakeFiles/serial.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

helloguys1913, 2017-11-01:
$ make
[  9%] Automatic moc for target my_project
[  9%] Built target my_project_automoc
[ 18%] Automatic moc for target serial
[ 18%] Built target serial_automoc
[ 63%] Built target serial
[ 72%] Linking CXX executable my_project.exe
CMakeFiles/my_project.dir/examples/dialog.cpp.o:dialog.cpp:(.text+0x790): undefined reference to `serial::list_ports()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/my_project.dir/build.make:151: my_project.exe] Error 1
make[1]: *** [CMakeFiles/Makefile2:69: CMakeFiles/my_project.dir/all] Error 2
make: *** [Makefile:84: all] Error 2

sirop, 2017-11-16:
This is rather a Qt question.

wjwwood, 2018-01-13:
This usually is because you're not linking against the serial library.

I'm going to close this because I believe the issue has been addressed. Feel free to continue to comment and I can reopen if necessary.

closed 2018-01-13 by wjwwood