wjwwood/serial #163: catkin package 'serial' is not listed as a build dependency in the package.xml

DimitrisKld, 2017-11-02:
Hello!
I am trying to build a catkin package which includes the library serial  but i receive this error 

> **CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:196 (message):
  catkin_package() the catkin package 'serial' has been find_package()-ed but
  is not listed as a build dependency in the package.xml
  Call Stack (most recent call first):
  /opt/ros/kinetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  thesis/CMakeLists.txt:25 (catkin_package)**

Could anyone help?Thanks in advance!

wjwwood, 2017-11-02:
Please ask questions like this on https://answers.ros.org, see: http://wiki.ros.org/Support

> the catkin package 'serial' has been find_package()-ed but
is not listed as a build dependency in the package.xml

Is `serial` listed as a build depend in your `package.xml`? I think the error message is quite clear on that suggestion.

closed 2017-11-02 by wjwwood

DimitrisKld, 2017-11-04:
Yes, you are right! It was obvious and i forgot to close this issue.