wjwwood/serial #170: Unresolved symbol error visual studio 2017

Gunlek, 2018-01-13:
Everytime i try to compile with visual studio 2017, i'm getting in trouble with an unresolved symbol error (LNK2019)... How can I fix that ?

wjwwood, 2018-01-13:
Please post the full build output, without it there is no context for me to deduce what the issue might be.

Gunlek, 2018-01-14:
Of course, here's the build output I get:

1>------ Début de la génération : Projet : SerialTest, Configuration : Debug x64 ------
1>Main.obj : error LNK2019: unresolved external symbol "class std::vector<struct serial::PortInfo,class std::allocator<struct serial::PortInfo> > __cdecl serial::list_ports(void)" (?list_ports@serial@@YA?AV?$vector@UPortInfo@serial@@V?$allocator@UPortInfo@serial@@@std@@@std@@XZ) referenced in main function
1>PATH_TO_PROJECT\x64\Debug\SerialTest.exe : fatal error LNK1120: 1 externes non résolus
1>Génération du projet "SerialTest.vcxproj" terminée -- ÉCHEC.
========== Génération : 0 a réussi, 1 a échoué, 0 mis à jour, 0 a été ignoré ==========

I'm french and so, output is french, I translated errors in english for you

wjwwood, 2018-01-15:
Thanks, typically, this kind of error will occur when either you have not linked against the serial library (not sure how to express that in Visual Studio) or in this specific linker error perhaps you're building [`serial.cc`](https://github.com/wjwwood/serial/blob/17e3223e6897a73479b1f7e1033086f9673ae4b1/src/serial.cc#L1) yourself, but have excluded the ["`list_ports`"](https://github.com/wjwwood/serial/tree/17e3223e6897a73479b1f7e1033086f9673ae4b1/src/impl/list_ports) part of the library.

klukander, 2021-04-16:
Hi, I ran into this with VS2019 and banged my head against the wall for quite some time with the typical visual studio linking things all in order. I then noticed that this happens with two functions in serial requiring registry access:

Error	LNK2019	unresolved external symbol __imp_RegCloseKey referenced in function "class std::vector<struct serial::PortInfo,class std::allocator<struct serial::PortInfo> > __cdecl serial::list_ports(void)" (?list_ports@serial@@YA?AV?$vector@UPortInfo@serial@@V?$allocator@UPortInfo@serial@@@std@@@std@@XZ)

Error	LNK2019	unresolved external symbol __imp_RegQueryValueExW referenced in function "class std::vector<struct serial::PortInfo,class std::allocator<struct serial::PortInfo> > __cdecl serial::list_ports(void)" (?list_ports@serial@@YA?AV?$vector@UPortInfo@serial@@V?$allocator@UPortInfo@serial@@@std@@@std@@XZ)

And then from a random stackoverflow response I already lost, found out that these need to be linked against
Advapi32.lib

Adding that to my project's linked libraries seems to solve the issue. 

I have no previous experience on this, and I don't quite understand why compiling the serial library itself, and the test_serial project included don't have an issue with this (the test runs fine).

Charlie-Gallie, 2024-06-03:
Apologies to add to such an old thread, but if anybody else encounters this issue even with the files included in your project and linked to Advapi32.lib, try changing the .cc file extensions to .cpp.

My version of Visual Studio doesn't compile .cc files by default. Changing them to .cpp fixed it.