wjwwood/serial #173: make failed

xxinine, 2018-01-24:
I have run "git clone git://github.com/wjwwood/serial.git" and now have the serial folder. cd into the serial folder. "make" produces this error:

cd build && cmake -DCMAKE_INSTALL_PREFIX=/tmp/usr/local ..
CMake Error at CMakeLists.txt:5 (find_package):
  By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "catkin", but
  CMake did not find one.

  Could not find a package configuration file provided by "catkin" with any
  of the following names:

    catkinConfig.cmake
    catkin-config.cmake

  Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
  "catkin_DIR" to a directory containing one of the above files.  If "catkin"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/yan/Documents/serial/build/CMakeFiles/CMakeOutput.log".
Makefile:32: recipe for target 'serial' failed
make: *** [serial] Error 1

smkanadl, 2018-01-24:
As the Readme states, catkin is a required dependency to build the serial lib. There are several issues around this topic as (from a code point of view) catkin is not really required (#111 #123 #133 #135). @wjwwood said [here](https://github.com/wjwwood/serial/pull/111#issuecomment-357462369) that a pure CMake support will come with the next version of serial.

thangduong, 2018-07-14:
even with catkin, it doesn't make:
-- catkin 0.6.16
Traceback (most recent call last):
  File "/data/robo/serial/catkin_generated/generate_cached_setup.py", line 12, in <module>
    from catkin.environment_cache import generate_environment_script
ImportError: No module named 'catkin'

During handling of the above exception, another exception occurred:

It seems the generated py file has a problem.  

if os.path.exists(os.path.join('/usr/share/catkin/cmake', 'catkinConfig.cmake.in')):

probably should be:

if os.path.exists(os.path.join('/usr/share/catkin/cmake', 'catkinConfig.cmake')):

Maybe it worked with a different version of catkin.  Seems overkill to have all these packages just for serial port.