wjwwood/serial #175: ImportError: No module named catkin.environment_cache

RdlP, 2018-04-25:
When I try to run the `make` command y get the following error:
```ImportError: No module named catkin.environment_cache```

This is the output of ```make``` command

```
 $ cmake -Dcatkin_DIR=/opt/ros/indigo/share/catkin/cmake/ ..
-- The C compiler identification is GNU 5.4.1
-- The CXX compiler identification is GNU 5.4.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/Angel/serial/build/devel
-- Using CMAKE_PREFIX_PATH:
-- Found PythonInterp: /usr/bin/python (found version "2.7.6")
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/Angel/serial/build/test_results
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
Traceback (most recent call last):
  File "/home/Angel/serial/build/catkin_generated/generate_cached_setup.py", lin                                                                      e 20, in <module>
    from catkin.environment_cache import generate_environment_script
ImportError: No module named catkin.environment_cache
CMake Error at /opt/ros/indigo/share/catkin/cmake/safe_execute_process.cmake:11                                                                       (message):
  execute_process(/usr/bin/python
  "/home/Angel/serial/build/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/all.cmake:186 (safe_execute_process)
  /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/Angel/serial/build/CMakeFiles/CMakeOutput.log".
See also "/home/Angel/serial/build/CMakeFiles/CMakeError.log".
```

How can fix the error?

thangduong, 2018-07-14:
I am seeing the same problem.   I have catkin installed

aviallon, 2018-08-05:
Same problem here.