wjwwood/serial #178: High delay in communication between Teensy 3.2 and Tegra TX1

richrdcm, 2018-05-15:
Hi guys!

I hope you can help me, I already struggled with this a lot of time and I do not find a solution.

First I was trying to send a bunch of data but now I am just simply sending the yaw value of an IMU.
The problem is that when I run the serial rx node on ROS and I echo the topic, the data is shown with a really big delay, it seems like is being stored in some buffer, the delay is between 500 ms and 1 sec on the Tegra Tx1. This means, I start moving the Imu and after the delay, the data starts moving, I stop moving the imu and the data still comes and stops after a delay. I am only running the node to receive the data through serial. 

The strange thing is that it does not happen on my laptop or PC running the same node and teensy program, and when I read the serial with minicom on the Tegra everything is fine, so it is not my hardware.

I already tried:
-Using rosserial with pyserial versions 2.5, 2.6 and 3.0.1 and limiting the rate of the published data from 30 hz to 100 hz.
-Different baudrates from 9600 to 500000
-Setting values to the serial port with setserial such as baud_rate, closing_wait, close_delay and divisor.

At this moment I am pretty sure it is not my code but anyway here is the code versión I am using in c++:

Header:

```
#include <ros/ros.h>
#include <ros/message_operations.h>
#include <std_msgs/String.h>
#include <std_msgs/Int16.h>
#include <std_msgs/UInt16.h>
#include <std_msgs/Bool.h>
#include <std_msgs/Float32.h>
#include <sensor_msgs/Imu.h>
#include <string>
#include <iostream>
#include <cstdio>
#include <unistd.h>
#include "serial/serial.h"
#include <sstream>
#include <ros/console.h>
#include <sstream>
#include <thread>
#include <atomic>

using std::string;
using std::exception;
using std::cout;
using std::cerr;
using std::endl;
using std::vector;
typedef int16_t speed_MMpS_t;

class imu
{
  private:
    //! Node handle in the private namespace
    ros::NodeHandle priv_nh_;
    ros::NodeHandle nh_;
    ros::Publisher pub_imu;
    std::string serial_port_;
    int baud_rate_;
    std::string result;
    size_t bytes_wrote;

    serial::Serial my_serial;
    std::string readline();
    void processValues();
    std::thread mThread;

    std::atomic<bool> mIsRunning {true};

    std_msgs::Float32 imu_msg = std_msgs::Float32(); 	

  public:

    imu(ros::NodeHandle nh);
    ~imu();
    void init();
    void my_sleep(unsigned long milliseconds);
};
```

cpp:

```
#include <imu/imu.h>
#include <thread>
#include <atomic>

imu::imu(ros::NodeHandle nh): nh_(nh), priv_nh_("~"), my_serial("/dev/ttyACM0",115200, serial::Timeout::simpleTimeout(1000))
{
  result="";
  pub_imu = nh_.advertise<std_msgs::Float32>(nh_.resolveName("/imu"), 1);
  priv_nh_.param<std::string>("arduino_serial_port", serial_port_, "/dev/ttyACM0");
  priv_nh_.param("arduino_baud_rate", baud_rate_,115200);
  my_serial.close();
  my_serial.setPort(serial_port_);
  my_serial.setBaudrate(baud_rate_);
  my_serial.open();

  mThread = std::thread([this]() { while(mIsRunning) {
    my_sleep(1);
    processValues();
  }});

}

imu::~imu() {
  mIsRunning = false;
  mThread.join();
}

void imu::processValues() {

  auto result = readline();
  //cout << "Yaw: " << result <<"\n";
  if(result == "Y")
  {
    auto value = readline();
    std::stringstream ss(value);
    ss >> imu_msg.data;
  }
  
  pub_imu.publish(imu_msg);

}

void imu::my_sleep(unsigned long milliseconds) {
    usleep(milliseconds*1000); // 100 ms
}

void imu::init()
{
  try
    {
      ROS_INFO("imu::Is the serial port %s open?",serial_port_.c_str());
      //cout << my_serial.getBaudrate() << endl;
	  if(my_serial.isOpen())
	    ROS_INFO("imu::Yes.");
	  else
	    ROS_ERROR("imu::No.");
    }
    catch(const std::exception& e)
    {
      	ROS_ERROR("imu::could not find serial port");
    }
}

std::string imu::readline()
{
  std::string result = "";
  uint8_t c = '\0';
  // read until new line terminates result string
  while (c  != '\n')
  {
      my_serial.read(&c, 1);
      if(c != '\n' and c != '\r')
      {
      result += char(c);
      }
  }
    return result;
}
```

node:
```
#include <imu/imu.h>

int main(int argc, char **argv)
{
  ros::init(argc, argv, "imu_node");
  ros::NodeHandle nh;
  ros::Rate loop(100);
  imu node(nh);
  ros::spinOnce();
  node.init();
  while(ros::ok())
  {
    //node.run(1000);
    ros::spinOnce();
    //loop.sleep();
  }
  return 0;
}```

Arduino:
```
#include <ros.h>
#include <Streaming.h>
#include <Servo.h>
#include "Imu.h"

Imu imu = Imu(8);

void setup()
{
  Serial.begin(115200);
  imu.initialize();
}

void loop() 
{ 
  Serial  << "Y"<< endl << imu.ypr[0] << endl;
}
```

I am out of Ideas, can you guys suggest me somewhere to look for a possible guilty of this delay.

Thanks!.

scott-eddy, 2018-05-17:
[Flush the serial input buffer](https://github.com/wjwwood/serial/blob/master/src/serial.cc#L359) prior to calling `read`.  This will ensure that the data you're reading is the latest measurement.

richrdcm, 2018-05-17:
Thanks! This made the trick, but I am curious to know why is it that on my computer worked without flushing the Input buffer, is it related to the size of the buffer on the Tegra?.