wjwwood/serial #182: Bizzare error when running program built serial as dependency

aviallon, 2018-08-05:
```
$ valgrin ./Oscilloscope
==5140== Memcheck, a memory error detector
==5140== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==5140== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==5140== Command: Release/Oscilloscope
==5140== 
**terminate called after throwing an instance of 'serial::IOException'
  what():  IO Exception (2): No such file or directory, file /home/aviallon/git/serial-git/src/serial/src/impl/unix.cc, line 151.**
==5140== 
==5140== Process terminating with default action of signal 6 (SIGABRT): dumping core
==5140==    at 0x6D7186B: raise (in /usr/lib/libc-2.27.so)
==5140==    by 0x6D5C40D: abort (in /usr/lib/libc-2.27.so)
==5140==    by 0x6260449: __gnu_cxx::__verbose_terminate_handler() [clone .cold.1] (vterminate.cc:95)
==5140==    by 0x62669D5: __cxxabiv1::__terminate(void (*)()) (eh_terminate.cc:47)
==5140==    by 0x6266A12: std::terminate() (eh_terminate.cc:57)
==5140==    by 0x6266C47: __cxa_throw (eh_throw.cc:95)
==5140==    by 0x4E3E7EA: ??? (in /usr/lib/libserial.so)
==5140==    by 0x4E425F8: serial::Serial::SerialImpl::SerialImpl(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long, serial::bytesize_t, serial::parity_t, serial::stopbits_t, serial::flowcontrol_t) (in /usr/lib/libserial.so)
==5140==    by 0x4E3F010: serial::Serial::Serial(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int, serial::Timeout, serial::bytesize_t, serial::parity_t, serial::stopbits_t, serial::flowcontrol_t) (in /usr/lib/libserial.so)
==5140==    by 0x116350: Oscilloscope (oscilloscope.h:46)
==5140==    by 0x116350: main (main.cpp:533)
==5140== 
==5140== HEAP SUMMARY:
==5140==     in use at exit: 51,993 bytes in 294 blocks
==5140==   total heap usage: 346 allocs, 52 frees, 165,980 bytes allocated
==5140== 
==5140== LEAK SUMMARY:
==5140==    definitely lost: 0 bytes in 0 blocks
==5140==    indirectly lost: 0 bytes in 0 blocks
==5140==      possibly lost: 1,786 bytes in 22 blocks
==5140==    still reachable: 50,207 bytes in 272 blocks
==5140==                       of which reachable via heuristic:
==5140==                         newarray           : 1,536 bytes in 16 blocks
==5140==         suppressed: 0 bytes in 0 blocks
==5140== Rerun with --leak-check=full to see details of leaked memory
==5140== 
==5140== For counts of detected and suppressed errors, rerun with: -v
==5140== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

I really don't understand what happens, I tried to build the package myself, but failed with same bug as #175. I think there is a problem within serial itself (which is why I post that here).

kamocuvao, 2018-08-07:
I have exactly the same error, but I did not update the serial library.

`Unhandled Exception: IO Exception (2): No such file or directory, file /path/to/repo/src/impl/unix.cc, line 151.`
I'm running in a VirtualBox (5.2.12 r122591) under Debian Sid (fully upgraded).

wjwwood, 2018-08-07:
In both cases, the file you're asking Serial to open does not exist. See:

https://github.com/wjwwood/serial/blob/5a354eaab7c064a155857d377a2fa3f0f8390c34/src/impl/unix.cc#L139-L151

closed 2018-08-07 by wjwwood