wjwwood/serial #201: vendor and product read interface if available

rahuldeo2047, 2019-04-11:
In case the serial is attached to the system and if a device which can provide vendor or product information; a method will allow users to enumerate all the serials and select the specified serial connection which has the given vendor and product ID associated.
Information can be availed as follows:

VendorID
ProductID
DeviceID
iSerialNumber

rahuldeo2047, 2019-04-11:
OK found usb_sysfs_friendly_name

will check it first.

wjwwood, 2021-06-10:
Might be useful if it can be made to be portable. Closing for now, feel free to comment and I can reopen if needed.

closed 2021-06-10 by wjwwood

rahuldeo2047, 2021-06-11:
I will get back to check. Unable to recall currently. Will check usb_sysfs_friendly_name as well. Aimed to select one serial interface when multiple same vid and pid connected and aimed to see if rest of the two did or serial number differed. And remained identifiable everytime. Thanks for the library.