wjwwood/serial #202: Not detecting all serial ports on Windows

tehKaiN, 2019-04-16:
So far everything worked fine for me in my app, but apparently this library doesn't detect virtual serial ports exposed by some 3rd party apps, namely VSPE virtual serial port emulator.

I've noticed that Realterm doesn't have this issue, so I peeked into its sources and it detects serial ports using registry path `HKEY_LOCAL_MACHINE\HARDWARE\DEVICEMAP\SERIALCOMM`.

I think that VPSE may be doing something wrong in adding its serial ports to Windows, but perhaps it'd be wise to expand current serial port list code by looking for additional ports in this registry path?

zhouscd, 2019-06-25:
We should use GUID_DEVINTERFACE_COMPORT instead of GUID_DEVCLASS_PORTS to scan serial ports on Windows
see: [https://docs.microsoft.com/en-us/windows-hardware/drivers/install/guid-devinterface-comport](url)

wjwwood, 2021-06-10:
A pull request would be welcome. A potentially related pull request was recently merged: https://github.com/wjwwood/serial/pull/179