wjwwood/serial #203: Issue when used with ncurses.

Odog1996, 2019-04-25:
Really random issue but when compiled through gcc version 5.4.0 20160609 (Ubuntu 5.4.0-6ubuntu1~16.04.11), serial/serial.h must be included in the program before ncurses.h otherwise the following build error occurs:

`/usr/local/include/serial/serial.h:471:45: error: macro "timeout" passed 5 arguments, but takes just 1`
`                     write_timeout_multiplier);`

HRex39, 2021-07-01:
hello，I wonder if you solve this problem and how.
Thanks a lot.

HRex39, 2021-07-02:
I tried to skip the macro and just call the function like this:  
`serial/serial.h:471`
```
Timeout (timeout)(inter_byte_timeout, read_timeout_constant,
                    read_timeout_multiplier, write_timeout_constant,
                    write_timeout_multiplier);
    return setTimeout(timeout);
```
The Package seems then compile successfully.