wjwwood/serial #205: Inappropriate ioctl for device on rapberry pi raspbian 8 jessie

mikelaltadi, 2019-05-07:
When I use available in  serial port ttyAMA0
 in rapberry pi
 raspbian 8 jessie

using wjwwood/serial for Linux
I receive the following error:
IO Exception (25): Inappropriate ioctl for device,
file /src/unix.cpp, line 488.

Serial::SerialImpl::available ()
{
  if (!is_open_) {
    return 0;
  }
  int count = 0;
  if (-1 == ioctl (fd_, TIOCINQ, &count)) {
      THROW (IOException, errno);
  } else {
      return static_cast<size_t> (count);
  }
}

ComeBackTo2016, 2020-04-30:
I have the same Question on rpi3bplus based on ubuntu server 18.04, linux 4.15. 
Did you solve this problem?

GXTX, 2020-05-08:
Also running into this on a rPi.

wjwwood, 2021-06-10:
I don't have an rpi to try this on, and normal linux doesn't seem to have this (for me), so I'll have to close this until someone can come up with a solution. If any new information comes up, or a solution is found, please comment here and I can reopen.

closed 2021-06-10 by wjwwood

GXTX, 2021-06-11:
If you can't test it doesn't it seem reasonable to have the issue open if users *are* running into this? I don't see the point in closing it.