wjwwood/serial #212: MacOS high baud rates

thestumbler, 2019-09-07:
Has anyone been successful getting higher baud rates, say 1M and above, working?  I've tested my hardware and it's working, using a terminal emulator or python.  But not with libserial.  I max out at 921600.

MacOS version 10.14.6

thestumbler, 2019-09-09:
One additional data point.  I was initially lulled into thinking it was working because the loopback test worked great.  It was only when I tried to actually communicate with another device at these higher baud rates did I realize the problem.  Regarding the loopback, it seems that the code *thinks* that is successfully configured the higher baud rate, but in actuality it didn't and the port defaulted to 9600 baud on my system.  The loopback testing passed, but it was not running at the desired speed.

I also tried briefly to apply the patch from Picocom as suggested by Mike Purvis in issue [#167](https://github.com/wjwwood/serial/pull/167) but without success.  To get my project going, I've switched to Sigrok's libserialport for the time being.  But I want to also get this library working eventually, as well.

wjwwood, 2021-06-10:
I recently merged https://github.com/wjwwood/serial/pull/218 which might fix this.

closed 2021-06-10 by wjwwood