wjwwood/serial #217: can write but can not read any strings

guoxinzhan, 2020-03-20:
Is the serial port open? Yes.
Timeout == 1000ms, asking for 1 more byte than written.AT9600
Iteration: 0, Bytes written: 2, Bytes read: 0, String read:
Iteration: 1, Bytes written: 2, Bytes read: 0, String read:
Iteration: 2, Bytes written: 2, Bytes read: 0, String read:
Iteration: 3, Bytes written: 2, Bytes read: 0, String read:
Iteration: 4, Bytes written: 2, Bytes read: 0, String read:

shiwenwang, 2020-09-09:
Is the serial port open? Yes.
Timeout == 1000ms, asking for 1 more byte than written.
Iteration: 0, Bytes written: 6, Bytes read: 6, String read: AT\r\n
Iteration: 1, Bytes written: 6, Bytes read: 6, String read: AT\r\n
Iteration: 2, Bytes written: 6, Bytes read: 6, String read: AT\r\n
Iteration: 3, Bytes written: 6, Bytes read: 6, String read: AT\r\n
Iteration: 4, Bytes written: 6, Bytes read: 6, String read: AT\r\n
Iteration: 5, Bytes written: 6, Bytes read: 6, String read: AT\r\n

Reading worked, but theresult is unexcepted for me. It should be `\r\nOK`.

wjwwood, 2021-06-10:
Seems like an issue with the device that is connected, not the library? Closing.

closed 2021-06-10 by wjwwood