wjwwood/serial #222: Noetic Release

tonybaltovski, 2020-05-12:
Could you please do a release for Noetic?

doisyg, 2020-05-25:
Yes, that would be nice as many drivers rely on this

gmsanchez, 2020-06-22:
Yes, please

Juba19, 2020-09-02:
Agree, I hope soon ! It will be a helpful tool for NOETIC. Thank you

droemer7, 2020-09-06:
Seconding this as well, but I suppose it can be built from source and should work?

doisyg, 2020-09-07:
> Seconding this as well, but I suppose it can be built from source and should work?

Yes, working for me from src

Tobias-Fischer, 2020-10-13:
Ping @wjwwood 
Working for me fine from src, too.

joshnewans, 2020-12-07:
It would be great to have this available as a noetic release. It is working fine from source for me, but it is a dependency for my package which I would like to be able to distribute for noetic.

stwirth, 2021-01-13:
@wjwwood ping

julianraheema, 2021-01-20:
Any time soon for this package to be in distribute for noetic as ros package?

HappySamuel, 2021-02-15:
@Tobias-Fischer 

How do you get it work on Noetic with src? I managed to catkin_make successfully with the src, but when i run the program, it crashed. 

[image]

Best,
Samuel

Tobias-Fischer, 2021-02-16:
Hi @HappySamuel - it's impossible to tell from the screenshot. As the error message says, the serial port is already open. This seems more like a bug in the downstream code? I didn't need to modify anything in the code that I am using (um7).

lauralindzey, 2021-02-19:
@wjwwood bump =)

Tobias-Fischer, 2021-04-11:
Ping @wjwwood

HappySamuel, 2021-04-21:
Hi @Tobias-Fischer 

I found the root cause of that issue. It's because of my another package that caused the problem, they interfere each other if they're catkin_make under same workspace. If separated them into different workspace, and linked together, this issue will not persist. Previously on Ubuntu 16.04 and 18.04, i didn't face this issue, it's because i installed the serial package using sudo apt-get install, hence it's in different workspace (/opt/ros/version/...). Due to yet to have distribute version of serial package in ros noetic, this is the solution i applied.

Hope, there'll soon release noetic version of serial package. @wjwwood 

Best,
Samuel

HappySamuel, 2021-06-10:
Still waiting for noetic release...

SubaruArai, 2021-12-06:
If there is nobody doing it, I can make a pr to ros/rosdistro. @wjwwood can I make a pr to rosdistro/noetic?

SubaruArai, 2021-12-06:
@wjwwood I've made a draft, please approve if you feel ok.
https://github.com/ros/rosdistro/pull/31385

SubaruArai, 2021-12-06:
@wjwwood I made a draft pr. If you think it's fine, please accept.
https://github.com/ros/rosdistro/pull/31386

wjwwood, 2021-12-08:
I had to close it, I don't think you used bloom to release the repository, which is required. I was wanting to switch to pure CMake, away from catkin, and to do that I needed to fix CI with GitHub Actions, and so on and so forth. I haven't had time to do that, so I'll just release it as-is.

wjwwood, 2021-12-08:
https://github.com/ros/rosdistro/pull/31420 this will get merged soon prob

closed 2021-12-08 by wjwwood

SubaruArai, 2021-12-08:
@wjwwood My bad, thank you!