wjwwood/serial #229: More packaging: v8stdint.h clash

leamas, 2020-09-17:
The v8stdint.h header is used in multiple packages, and creates collisions in large namespaces like Debian. Furhermore, this file is not really needed on any linux systen, since these have stdint.h available. 

Enclosed patch drops v8stdint.h from package on hosts having stdint.h available, thus resolving this issue.

[0001-Avoid-using-v8stdint.h-unless-needed.patch.gz](https://github.com/wjwwood/serial/files/5238914/0001-Avoid-using-v8stdint.h-unless-needed.patch.gz)

wjwwood, 2020-09-17:
Please consider a pull request instead of a patch file: https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request

Also, `v8stdint.h` is used conditionally on Windows only, so it should never cause an issue with Debian:

https://github.com/wjwwood/serial/blob/cbcca7c83745fedd75afb7a0a27ee5c4112435c2/include/serial/v8stdint.h#L39

I'll close this since I cannot take a patch file (no git author, etc).

closed 2020-09-17 by wjwwood

leamas, 2020-09-17:
The problem is not the usage, the problem is the very existence of this file which typically goes int /usr/include, creating collisions with other packages. 

The patch is a git patch, you can use git  -am which brings you author, date  etc. I was admittedly lazy, the debian packaging lives on gitlab...

wjwwood, 2020-09-17:
I see, given this is a very background project for me, I still don’t see me taking it without a pull request. But the patch is useful for others that stumble here. Thanks!