wjwwood/serial #238: Removal of opened port not detected properly.

randohinn, 2021-02-23:
See https://github.com/Teaduskeskus-AHHAA/oskar_driver/blob/cf0a9f6b7cb7d4382fb9ac8d07a1e08d070df620/src/board_comms.cpp for how I'm using this serial library in ros/ubuntu18.04

I've developed a bit of a problem.. When transmitting, if I remove my serial port.. it will thrown a exception, in which serial_.isOpen will still be true. When I try to manually close, it will complain about calling terminate after a portnotopened exception. As you may see, I have a interval configure to reconnect, since the serial instance will not detect a proper disconnect, that reconnection attempt will not run. Are there any ways to make serial_isOpen properly detect removal of the serial device?

wjwwood, 2021-06-10:
The short answer is "I'm not sure". It has been long enough that I don't know what the expected behavior should be.

I know that in the past, Linux had some issues with hardware disconnects, e.g. write would report ready but the device was actually not connected:

https://github.com/wjwwood/serial/blob/33e5a31ab77011504c8e37d7e9c3b9fa4229e625/src/impl/unix.cc#L587-L591

But I don't know if this is related to your use case. Perhaps you can just catch the exception when you try to close it and keep going? But I'm honestly not sure.

zacgibson21, 2023-05-03:
I am running into the same issue on Windows where once a port is disconnected "isOpen" still returns true.