wjwwood/serial #240: Create new tag/release

anton-danielsson, 2021-03-27:
The latest release (1.2.1) is now  6 years old.
Would be great to have a new tag/release.

I'm planning to add `serial` to the Conan package manager: https://github.com/conan-io/conan-center-index/pull/4943
and it would be great to be able to point to a newer release from https://github.com/wjwwood/serial/releases.

VioletGiraffe, 2021-06-03:
I second this request, I'm still using the latest release from 2015. It's great to see that some work has been done recently (albeit still over a year ago), but I don't know which revision is considered stable and safe to use in production.

wjwwood, 2021-06-10:
I'm planning on doing a new release soon.

VioletGiraffe, 2021-06-10:
Thank you for your work and for the great library!

anton-danielsson, 2021-06-11:
Agreed! Thank you for a great library!

anton-danielsson, 2021-06-11:
Just FYI: The library is available on Conan Center: https://conan.io/center/serial

ThinkalVB, 2021-08-12:
I'm trying to add serial to the VCPKG - c++ package manager. If it was released, it would have been great.
https://github.com/microsoft/vcpkg/issues/19438