wjwwood/serial #243: Not working with Apple M1 / Silicon

benkuper, 2021-05-02:
Hello there,
and thank you for this wonderful lib !
I'm using it it in my project Chataigne : http://benjamin.kuperberg.fr/chataigne
Some users have reported that listing of the port on M1 laptops is not working.. other softwares are able to detect USB serial boards (ESP32, arduino boards), but listing on Apple M1 is just not working.
I can't verify for myself but it may be interesting to know if other people have problems or if it is an expected behaviour at all.

Thank you

wjwwood, 2021-06-10:
I haven't tried it (may have to finally get an M1 to test it), but if you have any more information about what's not working or the scenario that would be helpful. I just merged a pull request related to custom baud rates on macOS (https://github.com/wjwwood/serial/commit/a93fc844d9ff58d26c1bbd93cf3b799b57cd13ee), so maybe try master?

benkuper, 2021-06-13:
Hey, I don't have M1 as well (don't even have a mac, just VM)..
What I think it happening is that the list returned by serial::list_ports() is empty. 
At least from what I understand, the UI on my software is not showing USB serial port that are connected (and detected by other softwares like Arduino IDE).

benkuper, 2023-04-08:
Hello, update on that, it seems not working still and I could test with a Silicon :
My software was filtering VID/PID 0x0, it seems that all detected devices are returning vid / pid 0x0 on silicon.
At first I thought it was Rosetta's abstraction since my software was compiled for Intel only, but I recently made a native Silicon version and still the same problem.
I'll try to dig a bit more if it's a common problem shared by other software or if it comes from this library