wjwwood/serial #256: "Undefined reference to" problem

dvalnn, 2021-09-09:
I'm trying to compile a simple serial port list test using cmake.

The serial library was built according to the instructions on the main page (git clone, and all the make options)

The cmake code is as follows and apparently runs fine 

```
cmake_minimum_required(VERSION 3.1)
set(CMAKE_CXX_STANDARD 20)

project(serialTest)

set(serial_DIR "/tmp/usr/local/share/serial/cmake")

find_package(serial REQUIRED)
include_directories(${serial_INCLUDE_DIRS})

add_executable(serialTest test.cpp)

target_link_libraries(serialTest ${serial_LIBS})
```

However, calling make afterwards run into the undefined reference problems.
and gives me the following output:

```
❯ cmake ..
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dvalinn/prog/colAruco/testFolder/build
❯ make
[ 50%] Building CXX object CMakeFiles/serialTest.dir/test.cpp.o
[100%] Linking CXX executable serialTest
/usr/bin/ld: CMakeFiles/serialTest.dir/test.cpp.o: in function `main':
test.cpp:(.text+0x2a): undefined reference to `serial::list_ports()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/serialTest.dir/build.make:84: serialTest] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/serialTest.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```

the code i'm trying to compile is just a simple port check and its not giving me any warnings on vscode 
```
#include <stdlib.h>
#include <string>
#include <iostream>

#include <serial/serial.h>

#define MAX_DATA_LENGTH 255

char output[MAX_DATA_LENGTH];
char incoming[MAX_DATA_LENGTH];

int main(){
    std::vector<serial::PortInfo> availablePorts = serial::list_ports();

    for (auto port : availablePorts)
        std::cout << port.description << std::endl
                  << port.hardware_id << std::endl
                  << port.port << std::endl
                  << " ------- \n";
    // std::string port;

    // serial::Serial arduino(port);
}
```

I'm rather not very used to using external libraries such as these so any feedback is appreciated. 
I've also looked at issue #152 and while it might be similar it didn't help as I'm not using g++ to compile.

videh25, 2022-06-07:
Experiencing the same issue with catkin_make.

nyinyinyanlin, 2022-11-01:
I had a similar issue. But please have a look at these few steps. First, after `sudo make install`, I ran `sudo ldconfig`. This might help or might not.

Then, I tried to compile directly using g++ command.

`g++ src/serialTest.cpp -o build/serialTest -lserial`

This worked for me. Then I moved on to CMakeLists.txt. I tried to make everything simple.

```
cmake_minimum_required(VERSION 3.1)
set(CMAKE_CXX_STANDARD 20)

project(serialTest)

add_executable(serialTest ./src/serialTest.cpp)

target_link_libraries(serialTest serial)
```

This worked, too.