wjwwood/serial #258: macOS 11.3 Big Sur - port_info.description = n/a

stephanschulz, 2021-10-10:
I am using this library through this [openFrameworks.cc addon](https://github.com/bakercp/ofxSerial/issues/24) .
It has been working great previous macOS systems. But not with 11.3 on a Mac mini M1.
Now, connected Usb serial devices to not provide any `port_info.description` nor `port_info.hardware_id`

I am wondering if anyone knows how to make that feature work agin.
The related library part is here I believe:
https://github.com/wjwwood/serial/blob/main/src/impl/list_ports/list_ports_osx.cc#L239-L25

Thanks for any advice.

[image]
7

Paberu85, 2021-12-19:
Check if you can communicate with the device (it appears in sys report). Big Sur had some dramatic changes in the way it works with user level device drivers, in short, that have rendered many devices incompatible, as it's older device drivers became useless in Big Sur, that include CH340 usb-uart bridge, and many other devices, for which it's developers did not provided updated device drivers.

stephanschulz, 2021-12-19:
yes communication is still possible and works reliable. my problem is more about the lack of getting any `port_info.description`, which now returns `n/a`

Riuzakiii, 2022-01-21:
I have the same issue on Monterey 12.0.1, `port_info.description` and `port_info.hardware_id` return `n/a` which breaks my code because it is relying on `hardware_id` to identify the device.

Riuzakiii, 2022-01-21:
I have the same issue on a MacBook Pro with Apple M1, using the Conan package https://conan.io/center/serial. 

Both `port_info.description` and `port_info.hardware_id` return `n/a`, however it is working correctly on a Macbook Pro late 2013 with BigSur 11.6.2.

I have tried searching for information on what could have changed between BigSur and Monterey but came up empty handed.

benkuper, 2023-04-08:
For referencing, might be the same problem as #243

benkuper, 2023-04-08:
Tried the PR locally, it works :)