wjwwood/serial #273: Does it make sense to close the Serial object upon destruction ?

Hish15, 2022-08-30:
https://github.com/wjwwood/serial/blob/69e0372cf0d3796e84ce9a09aff1d74496f68720/src/serial.cc#L77

The current behavior is to delete an underlaying pointer, this does the job, but can lead to an exception on some cases.

Hish15, 2022-08-30:
I realized that `delete pimpl_` does call   `pimpl_` destructor which calls  `pimpl_::close`.

closed 2022-08-30 by Hish15

sudotamm, 2023-01-30:
It will crash if throw exception not been catched when `delete pimpl_`, how do you avoid this? @Hish15

Hish15, 2023-01-30:
Hello @sudotamm, I'm an user of this not a maintainer. 
What you say is right, the `delete pimpl_` calls `pimpl_::close` which is not noexcept...
You might want to open an issue/discussion