wjwwood/serial #284: Move serial to a ros-* org ?

moriarty, 2023-06-13:
> Any way we can get that merged back here and released? Or move serial to a ros-* org for Rover to potentially help maintain given their depencency on it (the best maintainers are users :smile: ).

_Originally posted by @SteveMacenski in https://github.com/wjwwood/serial/issues/204#issuecomment-686776892_

moriarty, 2023-06-16:
@wjwwood would you be open to adding maintainers to this project?

moriarty, 2023-06-20:
I would also suggest adding @tonybaltovski and myself as maintainers.

> We are looking into porting some drivers that depend on it:
>
> - https://github.com/ros-drivers/um7
> - https://github.com/ros-drivers/um6
> Thus, a release would be needed to bloom them as packages.

_Originally posted by @tonybaltovski in https://github.com/wjwwood/serial/issues/209#issuecomment-1538507835_

wjwwood, 2023-06-21:
Sorry, just getting to this.

I'd be fine with forking it to ros-drivers and archiving this repository, but I don't want to move it.

Also, I'd still like to see it renamed to avoid confusion a la https://github.com/ros-infrastructure/rep/pull/379, but that would be up to future maintainers.

closed 2023-06-21 by wjwwood

wjwwood, 2023-06-21:
To follow up on https://github.com/wjwwood/serial/pull/285#issuecomment-1601730630...

I just don't want to move it because then it doesn't show where it came from originally. Forking preserves the history of the repository in a more concrete way (in my opinion).

I figure any issues worth moving could be moved with the fork and then maintained there. Part of forking would naturally be triaging the issues/prs. Afterwards I would close any issues or prs, pointing them at the fork and then archive the repo.

Once moved, anyone can be made a maintainer without having to wait on me. We have processes in place to add/remove maintainers from the ros orgs.

wjwwood, 2023-06-21:
As for moving issues efficiently, you sadly cannot transfer issues across organizations, but we could consider making the fork on my account first, transferring issues, then moving the fork to ros-drivers. That's hacky, but might work. There's also https://github.com/gatewayapps/kamino, but I haven't investigated it thoroughly.

I know that seems like busy work, but I do think it's better to get this in a common space if so many people are depending on it and want to contribute to it.

I also believe that most of the issues are either out-of-date or invalid after so much time, and only a few are probably worth moving. So I don't think it's as bad as it sounds.

moriarty, 2023-06-26:
Okay, Forking is fine, should we also rename it? or keep the original name and then you'll archive your repo?

wjwwood, 2023-06-26:
I think you should rename the package just because serial is too generic and presumptuous. A la the updates to REP-144, but since you're moving it to ros-drivers org, I'm not sure what it should be named. Maybe ros_drivers_serial? And that should be the package name, c++ namespace, and header file/folder structure.