wjwwood/serial #288: ROS2 Support

joewong00, 2023-07-24:
I need to use this serial library in my ROS2 package. However, I cannot build and install this package because it has the dependency of `catkin`. Catkin is not supported by ROS2 now and I could not install it, how do I fix this issue

aserbremen, 2023-09-26:
Check out: https://github.com/RoverRobotics-forks/serial-ros2

clone it into your `src` folder of your colcon workspace and colcon build. Afterwards, you should be able to `find_package(serial REQUIRED)` and `ament_target_dependencies(target serial)` after sourcing the `setup.bash` of your colcon workspace.

jodle001, 2023-09-29:
It seems like the ros2 forks and there is a ros2 branch, that these are not pulling updates that are fixing bugs in main.  Is there a plan to make ros2 more supported at all?

jhrncar, 2024-02-18:
> Check out: https://github.com/RoverRobotics-forks/serial-ros2
>
> clone it into your `src` folder of your colcon workspace and colcon build. Afterwards, you should be able to `find_package(serial [...]

Hello, your reply helped me, thanks so much! But unfortunately, when I started my program using this library, i got the following error: 
[image]
Any ideas please on how to fix this?

Owhenthesaints, 2024-04-26:
A bit late, but this could possibly help (seeing as how it is permission denied) https://answers.ros.org/question/369846/ros2-start-node-with-sudo-rights/