wjwwood/serial #290: About setBaudrate()

ZJZ0405, 2023-09-13:
When I used a Baudrate of 100,000, I felt that it did not achieve the results I expected, and I suspected that it was using the default Baudrate (9600). @wjwwood

ZJZ0405, 2023-09-13:
Why do you have to use the switch to set the Baudrate and there is no prompt when it doesn't work???

lunarifish, 2024-07-31:
这个可能是一个历史遗留问题，靠termios头文件里预设的一组宏定义可以设置常用的标准波特率，但是要设置特殊波特率就会有很多种方法，比如自己改分频系数，这就要看硬件驱动的实现情况。据实测把`unix.cc`的430行开始这一段：
```c++
#elif defined(__linux__) && defined (TIOCSSERIAL)
    struct serial_struct ser;

    if (-1 == ioctl (fd_, TIOCGSERIAL, &ser)) {
      THROW (IOException, errno);
    }

    // set custom divisor
    ser.custom_divisor = ser.baud_base / static_cast<int> (baudrate_);
    // update flags
    ser.flags &= ~ASYNC_SPD_MASK;
    ser.flags |= ASYNC_SPD_CUST;

    if (-1 == ioctl (fd_, TIOCSSERIAL, &ser)) {
      THROW (IOException, errno);
    }
#else
```
改成这样：
```c++
#elif defined(__linux__) && defined (TCSETS2)
  struct termios2 tio2{};
  memcpy(&tio2, &options, sizeof(tio2));

  tio2.c_cflag &= ~CBAUD;
  tio2.c_cflag |= BOTHER;
  tio2.c_ispeed = baudrate_;
  tio2.c_ospeed = baudrate_;

  if (-1 == ::ioctl(fd_, TCSETS2, &tio2)) {
    THROW (IOException, errno);
  }
#else
```
可能可以解决问题。考虑到这个库的泛用性所以这个修改应该不适合提pr，自己改一下凑合用

btw国赛见 :)

ZJZ0405, 2024-07-31:
wow，rmer随处可见啊

获取 Outlook for iOS<https://aka.ms/o0ukef>

ZJZ0405, 2024-07-31:
有道理

closed 2024-07-31 by ZJZ0405