wjwwood/serial #296: make not working out of the box

vinayakyadav2709, 2024-01-07:
I have a 
Linux Mint 21.2 x86_64 system
after cloning and make command throwing this error

falcon_pc_s@Falcon-pc:~/Projects/pro_serial/serial$ make
cd build && cmake -DCMAKE_INSTALL_PREFIX=/tmp/usr/local ..
CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/falcon_pc_s/Projects/pro_serial/serial/build/devel
-- Using CMAKE_PREFIX_PATH: 
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python3
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/falcon_pc_s/Projects/pro_serial/serial/build/test_results
-- GTest is not available. Please install libgtest-dev to enable tests involving GTest.
-- GMock is not available. Please install libgmock-dev to enable tests involving GMock.
-- nosetests not found, Python tests can not be run (try installing package 'python3-nose')
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- catkin 0.8.10
-- BUILD_SHARED_LIBS is on
CMake Warning at /usr/share/catkin/cmake/test/gtest.cmake:159 (message):
  skipping gtest 'serial-test' in project 'serial' because gtest was not
  found
Call Stack (most recent call first):
  /usr/share/catkin/cmake/test/gtest.cmake:89 (_catkin_add_executable_with_google_test)
  /usr/share/catkin/cmake/test/gtest.cmake:37 (_catkin_add_g

ccajas, 2024-09-25:
Your pasted message seems to have cut off. If you also are getting this error:

`CMake Error at tests/CMakeLists.txt:3 (target_link_libraries):`

And you don't need to build a test, I found that commenting out this line in the **root** CMakeLists.txt can be a workaround for the build to work:

`add_subdirectory(tests)`

Right after `if(CATKIN_ENABLE_TESTING)`.

asdat3, 2024-09-27:
can someone provide the built lib? :/